OpenTracksApp/OpenTracks #786: Add sensor battery level warnings and readouts

gordonel, 2021-05-20:
**Is your feature request related to a problem? Please describe.**
I'd like to be notified if it's time to charge my devices or replace the batteries in advance.

**Describe the solution you'd like**
When starting an activity, OpenTracks should read the battery level if available and notify the user if it falls below a user-specified level. Also, the app can provide convenient readouts in the notifications drawer as extra info

**Describe alternatives you've considered**
reading battery levels through nRF connect before a workout

dennisguse, 2021-05-21:
Duplicate of #24, but way more detailed.

It is actually not that complicated anymore as passing data to the UI while recording became way easier.

gordonel, 2021-05-21:
Eeeh, sorry about that. Do you think it would be best to close one of the dups in favour of the other?

dennisguse, 2021-05-22:
@gordonel I really like your feature description; could you just add it to #24 and close this one?

gordonel, 2021-05-22:
Done

closed 2021-05-22 by gordonel