OpenTracksApp/OpenTracks #787: 3.18.1: App crashes after tap to intervals

vsevjednom-cz, 2021-05-21:
New error in last version 3.18.1.

**Describe the bug**
When I tap to intervals or chart in activity and after that tap to "back" without waiting for interval values app crashes.

**To Reproduce**
1. Go to some activity.
2. Tap to intervals.
3. Tap to "back" in your mobile.
4. App crashes.

`E/pentracks.debu: [qarth_debug:]  get PatchStore::createDisableExceptionQarthFile method fail.
E/AndroidRuntime: FATAL EXCEPTION: TrackDataHub
    Process: de.dennisguse.opentracks.debug, PID: 32277
    java.lang.NullPointerException: Attempt to invoke virtual method 'de.dennisguse.opentracks.stats.TrackStatistics de.dennisguse.opentracks.stats.TrackStatisticsUpdater.getTrackStatistics()' on a null object reference
        at de.dennisguse.opentracks.content.TrackDataHub.notifyTrackPointsTableUpdate(TrackDataHub.java:383)
        at de.dennisguse.opentracks.content.TrackDataHub.loadDataForListener(TrackDataHub.java:261)
        at de.dennisguse.opentracks.content.TrackDataHub.lambda$registerTrackDataListener$1$TrackDataHub(TrackDataHub.java:194)
        at de.dennisguse.opentracks.content.-$$Lambda$TrackDataHub$mbEXkhjouip_-npcoIfM1Jx00V4.run(Unknown Source:12)
        at android.os.Handler.handleCallback(Handler.java:907)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:216)
        at android.os.HandlerThread.run(HandlerThread.java:65)`

**Technical information**
 - Device: Honor KSA-LX9
 - OS: Android
 - Version 9.1.0

dennisguse, 2021-05-21:
I also saw a similar crash using v3.18.1

I guess unregistering in TrackDataHub regarding the ContenResolvers leads to a race condition.

dennisguse, 2021-05-21:
Just a quick fix.

@vsevjednom-cz Can you check if this fixes the issue for you?
It is not the nicest solution, but for now "okay".

vsevjednom-cz, 2021-05-21:
Yesss. It works amazing!

closed 2021-05-21 by vsevjednom-cz

dennisguse, 2021-05-22:
@vsevjednom-cz Thanks for testing it!

commit 6580bc41 (2021-05-21): TrackDataHub: prevent crash due to race condition.

Fixes #787.
PR #789 (2021-05-22): TrackDataHub: prevent crash due to race condition.
Fixes #787.
commit 1d782046 (2021-05-22): TrackDataHub: prevent crash due to race condition.

Fixes #787.