OpenTracksApp/OpenTracks #795: Allow rounding elevation scale domain

Crashillo, 2021-05-25:
**Is your feature request related to a problem? Please describe.**
In the image below, the elevation chart is barely meaningful, due to the scale begins at 0m, but the elevation mean was 750m-high roughly.
[image]

**Describe the solution you'd like**
Depicts a better range for the scales, based on the data boundaries, i.e. a slightly upper/lower to the maximum/minimum values.
Add a settings option in order to switch the scale behaviour

Crashillo, 2021-05-25:
Looking at my old tracklogs I noticed this feature was available back in late feb. Was it removed for any reason or just a bug?

dennisguse, 2021-05-26:
That was not implemented.
It was mentioned in #235, but you description is way nicer.

commit 07074ad2 (2022-01-02): TrackStatistics and ChartView: handle altitude and speed as optional with null as default value.

Then only the relevant part of the y-axis is shown.

Fixes #795.
closed 2022-01-02 by dennisguse