OpenTracksApp/OpenTracks #800: Crash while recording: clock went backwards?

dennisguse, 2021-05-28:
Reported via Google PlayStore.

I guess, we have a case in which the clock was adjusted while recording.
And the TrackStatistics then react with a RuntimeException.

Just happened once so far since we implemented using the _system clock_ instead of the _gps time_.

```
java.lang.RuntimeException: 
  at de.dennisguse.opentracks.stats.TrackStatistics.setStopTime (TrackStatistics.java:159)
  at de.dennisguse.opentracks.stats.TrackStatisticsUpdater.internalAddTrackPoint (TrackStatisticsUpdater.java:123)
  at de.dennisguse.opentracks.stats.TrackStatisticsUpdater.addTrackPoint (TrackStatisticsUpdater.java:108)
  at de.dennisguse.opentracks.services.TrackRecordingService.insertTrackPoint (TrackRecordingService.java:652)
  at de.dennisguse.opentracks.services.TrackRecordingService.newTrackPoint (TrackRecordingService.java:561)
  at de.dennisguse.opentracks.services.handlers.HandlerServer.lambda$sendTrackPoint$0$HandlerServer (HandlerServer.java:59)
  at de.dennisguse.opentracks.services.handlers.-$$Lambda$HandlerServer$u7UpyDYRGPit7CRfg_glQph5Wm8.run (Unknown Source:6)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:919)
```

dennisguse, 2021-05-28:
According to Android documentation, we may use some combination of `System.currentTimeMillis()` (once on track recording start) and later only use `elapsedRealtime()` to compute absolute timestamps ourselves.

https://developer.android.com/reference/android/os/SystemClock

velocitea, 2021-06-19:
My problem might be connected with this issue. My newest record seemed to work fine, but when I try to open the plots over time or distance, the app crashes consistently with the following exception

```
java.lang.RuntimeException: stopTime cannot be less than startTime: 2021-06-19T05:42:49.025Z 2021-06-19T05:42:49.016Z
       at de.dennisguse.opentracks.stats.TrackStatistics.setStopTime(TrackStatistics.java:159)
       at de.dennisguse.opentracks.stats.TrackStatisticsUpdater.internalAddTrackPoint(TrackStatisticsUpdater.java:127)
       at de.dennisguse.opentracks.stats.TrackStatisticsUpdater.addTrackPoint(TrackStatisticsUpdater.java:112)
       at de.dennisguse.opentracks.content.TrackDataHub.notifyTrackPointsTableUpdate(TrackDataHub.java:374)
       at de.dennisguse.opentracks.content.TrackDataHub.loadDataForListener(TrackDataHub.java:254)
       at de.dennisguse.opentracks.content.TrackDataHub.lambda$registerTrackDataListener$1$TrackDataHub(TrackDataHub.java:191)
       at de.dennisguse.opentracks.content.-$$Lambda$TrackDataHub$qVL0z0B5Tt90CX-W5z8AJJhv79s.run(Unknown Source:4)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:223)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```
I'm running LineageOS 18.1

dennisguse, 2021-06-19:
@velocitea Interesting.
Was the track recorded with OpenTracks?
As this crash should already be happening while the recording is running.
And if you recorded it with OpenTracks did you do something different than usual?

To fix a track: you can export and re-import it.
We reorder the trackpoints by time then (as of v3.18.3).

velocitea, 2021-06-19:
Yes it was recorded with OpenTracks and I didn't notice any problem before trying to display the charts. All stats seem to be correct like distance, time, pace etc. I am not sure if I did anything different, but actually i might have stopped and resumed. Also, I noticed the doc.kml is split in 3 tracks.

dennisguse, 2021-06-19:
:confused: 
That should not happen; we do the same computation while recording as well as for displaying the charts.
So, the crash should also happened while recording.
While recording this happening in TrackRecordingService (while recording to compute the TrackStatistics for the DB) and TrackDataHub for the UI (recording and afterwards; just used for charts).
Or we have a different problem...

(not that well documented) one gx:track in KML is a segment in OpenTracks.
Segments are created by pause/resume or too much distance to previous location (default 200m).

@velocitea if may share the track with me via email to dennis.guse@alumni.tu-berlin.de
Don't have too much time, but sounds like an interesting problem :)

commit f925b775 (2022-01-06): Bugfix: use monotonic clock.

Fixes #800.
commit ebc131f9 (2022-01-06): Bugfix: use monotonic clock.

Fixes #800.
commit 0e882556 (2022-01-06): Bugfix: use monotonic clock.

Fixes #800.
commit 48d50e9c (2022-01-06): Bugfix: use monotonic clock.

Fixes #800.
closed 2022-01-15 by dennisguse