OpenTracksApp/OpenTracks #805: Question: where is the default setting?

vsevjednom-cz, 2021-06-04:
Where are saved default settings? I can't find them. I want to change some default values.
And I need to change the default filename for example - I want to use id of user saved in sharedPref. I can't find where is generating of the filename.

dennisguse, 2021-06-04:
I guess, you are looking for:
* https://github.com/OpenTracksApp/OpenTracks/blob/01d83deefd115cf351ba1089a5a269d9742e2b6c/src/main/java/de/dennisguse/opentracks/util/TrackNameUtils.java

and the config options:
* https://github.com/OpenTracksApp/OpenTracks/blob/main/src/main/res/values/settings.xml
* https://github.com/OpenTracksApp/OpenTracks/blob/01d83deefd115cf351ba1089a5a269d9742e2b6c/src/main/java/de/dennisguse/opentracks/util/PreferencesUtils.java#L38

dennisguse, 2021-06-04:
@vsevjednom-cz Whats actually your plan?
Some user specific upload to some server based upon the file name?

vsevjednom-cz, 2021-06-04:
Yes, yes.
I've got idea to create better app + web than Strava 🤩. I created some [web interface](https://jedu.vsevjednom.cz/activity/2629) with [OpenLayers](https://openlayers.org/) and [ol-ext extension](https://viglino.github.io/ol-ext/) (amazing!!!) last year.

I'm trying to learn Kotlin/Java. GPS tracking at backgroud not works in Kotlin (I don't know how to do it). Your OpenTracks app is amazing! So I'm creating login and upload GPXs in app. It's fun for me 🥳.

This is not problem, I hope 🥴

I'll try your advices ⬆️ over the weekend.

dennisguse, 2021-06-04:
Looks nice!
An alternative option would be: share the tracks from within OpenTracks via e-Mail with your system. Depending on the sender's address, you could import it to the correct account.
Not really secure (as emails might not be encrypted), but as a prototype that should work.

PS: Implementing GPS tracking just requires a foreground service.
Getting the data is simple, but then it starts to get complicated: storing and exporting/importing data. Just too many details.

PPS: Kotlin has some nice features, but the trouble is usually Android and not the lack of language features in Java :)

vsevjednom-cz, 2021-06-06:
I decided save some users token to the gpx content, not to the filename. Example:
`printWriter.print("token=\"" + token + "\" ");`

I've got this: `public class LoginActivity extends AbstractActivity {` with method for get some saved token:
`    public String getValue(String name){
        SharedPreferences sharedPref = getPreferences(Context.MODE_PRIVATE);
        String value = sharedPref.getString(name, "");
        return value;
    }`

But i can't call this method from GPXTrackExporter. There's a problem because I'm calling non static function from static content. I can change method to static but there's not possible to call getPreferences in static method. I'm confused 🤪. I'm Googling hours and hours. Maybe I must save preferencer other way..?

vsevjednom-cz, 2021-06-06:
I decided save some user's token to the gpx content, not to the filename. Example:
`printWriter.print("token=\"" + token + "\" ");`

I've got this: `public class LoginActivity extends AbstractActivity {` with method for get some saved token:
`    public String getValue(String name){
        SharedPreferences sharedPref = getPreferences(Context.MODE_PRIVATE);
        String value = sharedPref.getString(name, "");
        return value;
    }`

But i can't call this method from GPXTrackExporter. There's a problem because I'm calling non static function from static content. I can change method to static but there's not possible to call getPreferences in static method. I'm confused 🤪. I'm Googling hours and hours. Maybe I must save preferencer other way..?

vsevjednom-cz, 2021-06-06:
I decided save some user's token to the gpx content, not to the filename. Example:
`printWriter.print("token=\"" + token + "\" ");`

I've got this: `public class LoginActivity extends AbstractActivity {` with method for get some saved token:
```
    public String getValue(String name){
        SharedPreferences sharedPref = getPreferences(Context.MODE_PRIVATE);
        String value = sharedPref.getString(name, "");
        return value;
    }
```

But i can't call this method from GPXTrackExporter. There's a problem because I'm calling non static function from static content. I can change method to static but there's not possible to call getPreferences in static method. I'm confused 🤪. I'm Googling hours and hours. Maybe I must save preferencer other way..?

dennisguse, 2021-06-06:
@vsevjednom-cz You can check the PreferencesUtils. That's how we access preferences in OpenTracks - almost all access is handled this way. 
And I guess, you should pass the value via GPXTrackExporter's constructor.

Reason: from a static method, you don't have access to properties of the object.
You only can access static properties and parameters. That's it.

IMHO (not meant to discourage you; just mentioning it)
Depending on how you introduce this feature, you need to keep in mind that your maintenance effort may be high.
For example, if we start cleaning up the internal classes (or rename etc), you may need to manually merge these conflicts - some parts of the code base are stable and others aren't.
Basically, you create a fork (as you also need to distribute your variant).

So, instead of modifying OpenTracks, I would recommend creating another app that can upload GPX/KML to your server if the user (a) exports the track using the automatic export feature or (b) using the share feature (manual track selection by user).
This way, we are just loosely coupled: OpenTracks just needs to export proper GPX/KML.
Also your users might want to use other applications like AAT, FitoTrack, or RunnerUp.

PS: If you really want access to the data, OpenTracks stores in it's database, you may use the Dashboard API - but would not recommend that for your use case.

vsevjednom-cz, 2021-06-08:
More applications are to too complicated for users. But you're right about maintenance (rename classes etc.). I created LoginActivity class with login and UploadActivity class with upload. It only looks into the folder and upload/delete all GPXs. Manual tap to the upload gpx files is worse than automatic but it's ideal in this situation.

Thank you for the tips! 🚀

vsevjednom-cz, 2021-06-08:
Last one question: Where can I change some default settings? For example - I want to turn on instant export and I want to set the GPX format.
I tried to search in settings.xml without success 🤪.

dennisguse, 2021-06-08:
```
    <string name="export_trackfileformat_default" translatable="false">KMZ_WITH_TRACKDETAIL_AND_SENSORDATA_AND_PICTURES</string>
```

For valid values check: `TrackFileFormat`.

vsevjednom-cz, 2021-06-09:
Yes! Can I set default export directory? For example something like this: getExternalFilesDir()+File.separator+"DIRECTORYFOREXPORT" (Android/data/*/files/DIRECTORYFOREXPORT). I can't see settings_default_export_directory_key_default 🧐. This is really last question, I promise 🤩.

dennisguse, 2021-06-09:
Not possible; we are using ScopedStorage and get the folder selection from the user.
It's in the SettingsActivity; there is an intent that is requesting it.
And we store the response.

You may do this directly after your LoginActivity is done.

dennisguse, 2021-06-09:
@vsevjednom-cz Don't worry about asking questions :)

vsevjednom-cz, 2021-06-10:
Thank you! Great idea about folder selection!

I'm trying to set the folder automatically in LoginActivity (it's useless without login) like this:

        String dir = getExternalFilesDir(null)+File.separator+"EXPORT"; //Android/data/*/files/EXPORT
        ...
        Uri directoryUri = Uri.parse(dir);
        PreferencesUtils.setDefaultExportDirectoryUri(sharedPreferences, this, directoryUri);

It only unsets folder for export.
And could you write me how to set `post_workout_export_enabled_default` to true and `export_trackfileformat_default` to GPX? I found `private void setExportTrackFileFormatOptions()` but there aren't input variables.

dennisguse, 2021-06-10:
That does not work (due to scoped storage): https://developer.android.com/about/versions/11/privacy/storage

This preference config is used to get the user approval for a directory.
This actually opens a directory chooser (Android OS) and returns us the URI to this folder.
And then we can write into it.
```
        <Preference
            android:key="@string/settings_default_export_directory_key"
            android:title="@string/settings_default_export_uri_title">
            <intent
                android:targetClass="de.dennisguse.opentracks.io.file.importer.DirectoryChooserActivity$DefaultTrackExportDirectoryChooserActivity"
                android:targetPackage="@string/applicationId" />
```

And to set the preferences: you can this via sharedPreferences as usual.
We did not implement those as they are only accessed via the SettingsActivity and Android is taking care of storing these infos.
`setExportTrackFileFormatOptions()` is only used to get the potential options that are presented to the user.

vsevjednom-cz, 2021-06-11:
I understand. Can I get only a value of export directory? I want to show a notification when directory is not set or is set differently.

I found this:
```
                Preference instantExportDirectoryPreference = findPreference(getString(R.string.settings_default_export_directory_key));
                instantExportDirectoryPreference.setSummaryProvider(preference -> {
                    DocumentFile directory = PreferencesUtils.getDefaultExportDirectoryUri(sharedPreferences, getApplicationContext());
                    //Use same value for not set as Androidx ListPreference and EditTextPreference
                    return directory != null ? directory.getName() : getString(R.string.not_set);
                });
```
But I can't find `findPreference` method.

vsevjednom-cz, 2021-06-12:
and... it doesn't matter which folder. User can set some directory and I can upload GPX files from this.  This is the reason why I need to get directory from preferences.

dennisguse, 2021-06-12:
Just check for `PreferencesUtils.getDefaultExportDirectoryUri(sharedPreferences, getApplicationContext());` for `null`, then the user has not yet set an export directory.
BUT the user might still revoke access to this folder from outside of OpenTracks.
If this happens, we still have the location stored in the preferences, but we cannot export something.

vsevjednom-cz, 2021-06-12:
Yes! This is it! I can refer people to settings and I can sync GPX files from a directory that people have set up. 
It's amazing!
😍 🚀

closed 2021-06-20 by vsevjednom-cz