OpenTracksApp/OpenTracks #808: Intervals doesn't work properly with TrackDataHub

rgmf, 2021-06-07:
I'm improving Intervals using TrackStatisticsUpdater to compute the intervals and I found that Intervals are re-created every time the IntervalsFragment is resumed. It doesn't make sense because we're using ViewModel with LiveData that retain the intervals. That's because TrackDataHub is resumed every time IntervalsFragment is resumed and then TrackDataHub sends all track points again and again.

It's potentially inefficient because if you change ten times from stats tab to intervals tab then intervals are re-computed ten times, for instance.

I'm working on it (https://github.com/OpenTracksApp/OpenTracks/tree/intervalsWithStatsUpdater%23703)

dennisguse, 2021-06-07:
Sounds good! 
Does this only affect the IntervalFragment?
I am pretty sure that the ChartFragment is doing it something very similar...

rgmf, 2021-06-07:
> I am pretty sure that the ChartFragment is doing it something very similar...

Yes, I think so too.

I'll check it out too.

commit 90e922b7 (2021-06-09): IntervalsFragment doesn't use TrackDataHub anymore. This improve intervals performance. Fixes #808.
commit 5d200e39 (2021-06-09): IntervalsFragment doesn't use TrackDataHub anymore. This improve intervals performance. Fixes #808.
commit 718117f6 (2021-06-09): IntervalsFragment doesn't use TrackDataHub anymore. This improve intervals performance. Fixes #808.
commit 838671e2 (2021-06-15): IntervalsFragment doesn't use TrackDataHub anymore. This improve intervals performance. Fixes #808.
closed 2021-06-15 by rgmf