OpenTracksApp/OpenTracks #814: The icon of OpenTracks is way too small on my Android 5

Altonss, 2021-06-10:
**Describe the bug**
The icon of OpenTracks is way too small on my Android 5 phone.

**To Reproduce**
1. Probably you first need a similar phone
2. Look an the app icon on the launcher

If applicable:
[image]

**Technical information**
 - Device: [Samsung Galaxy Core Prime]
 - OS: [Android 5.0.2]
 - Version [e.g. v3.18.3 from F-Droid]

dennisguse, 2021-06-10:
Yes, I removed the downward compatible PNGs and we now use the a vector graphic (hope to avoid problems for reproducible builds). Introduced in v3.18.3.

@Altonss Can you compile to test it on your device?

Altonss, 2021-06-10:
Thanks for your quick reply!
Ok you mean that there might be an issue with the vector graphic scaling on older devices?
What should I compile? A version with a bugfix? Sorry that I didn't fully understand your answer :)

dennisguse, 2021-06-10:
:)

Yes, it is a scaling issue that might affect devices pre-Android 8.0; afterwards we have an adaptive icon.
So, far included pre-rendered PNGs and I wanted to get closer to create reproducible builds.
And PNGs might be randomly modified while building the APK: https://f-droid.org/en/docs/Reproducible_Builds/
(and we save some APK size)

Commit: https://github.com/OpenTracksApp/OpenTracks/commit/b0575c0f5b961b3c9f3b2cf4ab1c69938ed53e57

Relevant file: https://github.com/OpenTracksApp/OpenTracks/blob/f5ec91f0eb8bfce892e9e5df93ddaf4842bae91a/src/main/res/drawable/ic_launcher.xml

But I cannot reproduce the issue in the emulator, so you may need to test it yourself.
We had a similar issue ages ago on Xioami, where the icon was too small.

dennisguse, 2021-06-10:
Link to old issue, but for adaptive icons on Xioami: https://github.com/OpenTracksApp/OpenTracks/issues/121

dennisguse, 2021-06-21:
I can also reproduce it on a LineageOS 17.1: the icon in the share menu is too small (within a large white circle).

dennisguse, 2021-06-24:
@Altonss Please re-open, if this does not solve the issue.

closed 2021-06-24 by dennisguse

Altonss, 2021-06-24:
Thanks a lot, I will wait for the next release to be available on F-Droid and give you my feedback :)

dennisguse, 2021-06-24:
@Altonss if you have a sec you can also try it directly ;)

[OpenTracksUpstream-releasePlayStore.zip](https://github.com/OpenTracksApp/OpenTracks/files/6710836/OpenTracksUpstream-releasePlayStore.zip)

Altonss, 2021-06-24:
On the left the icon before the bugfix and on the right with the bugfix, I think it's better now :)
[image]

But I noticed this bug also on both versions, that should be addressed in another issue. Thats the screen when opening the app:
[image]

dennisguse, 2021-06-24:
Thanks for testing and great that it works.
Always cool to fix something that one cannot test :)

PS: The splashscreen will stay that way.
Android (<6.0; <API23) renders the image to small and then upscales.
It is not nice, but not that bad.
Adding a version differentiation adds (IMHO) too much complexity.

Altonss, 2021-06-24:
Thanks for the info ;)

commit 26490f5a (2021-06-24): Launcher icon (API <26): does not have a background circle.
Rendering a multilayer image is not consistent (size).

Fixes #814.