OpenTracksApp/OpenTracks #816: Make F-Droid store listings translatable

dennisguse, 2021-06-15:
aka `fastlane/metadata/android/SOMETHING-SOMETHING/`
* full_description.txt
* title.txt
* short_description.txt

Out of scope: changelogs.

dennisguse, 2021-06-16:
Done: https://weblate.bubu1.eu/projects/OpenTracks/app-store-metadata-files/

dennisguse, 2021-06-16:
@mondlicht-und-sterne As head of translations, does this work?

pstorch, 2021-06-18:
If it works, I want to set it up for the OSMDashboard as well.

dennisguse, 2021-06-18:
Works like charm.

I just disabled the translations for the changelogs by setting them read-only.
I guess it is not worth the effort to translate those.

And almost not configuration needed: https://docs.weblate.org/_/downloads/en/weblate-3.11/pdf/
[image]

closed 2021-06-18 by dennisguse

mondlicht-und-sterne, 2021-06-18:
I'll take care of it from 28.6 after my holiday.

pstorch, 2021-06-18:
> I just disabled the translations for the changelogs by setting them read-only.

@dennisguse how did you do that?

dennisguse, 2021-06-18:
Like this; but I am not sure if this also applies to new files :man_shrugging: 

https://docs.weblate.org/en/latest/admin/addons.html

[image]

pstorch, 2021-06-18:
@dennisguse thanks. Maybe I'll keep them for now, because I realized I have translated changelogs in the OSMDashboard :laughing:

dennisguse, 2021-06-18:
I just wanted to avoid that people start translating them as the value is rather limited, but hey if somebody wants to why not ><

mondlicht-und-sterne, 2021-06-18:
> @mondlicht-und-sterne As head of translations, does this work?

i am on holiday now and i can translate it after 28.6.2021

dennisguse, 2021-06-19:
@mondlicht-und-sterne enjoy it ;)