OpenTracksApp/OpenTracks #817: Failing test: espressoDeleteTrackTest

dennisguse, 2021-06-15:
i get:
```
androidx.test.espresso.IdlingResourceTimeoutException: Wait for [de.dennisguse.opentracks.EspressoDeleteTrackTest$MyIdlingResource] to become idle timed out
```

[image]

PS: Please ignore EspressoCategoryUnitTest as this is a problem with localization setting.

rgmf, 2021-06-16:
I can reproduce the error in AVD API 22 but not in AVD API 28 and 29 :neutral_face:

In other way, in API 28 and API 29 EspressoDeleteTrackTest doesn't work because when it clicks on record button it doesn't move to TrackRecordingActivity and it continues in TrackListActivity.

dennisguse, 2021-06-17:
I did not check this yet. I will try to take a look at the weekend ;)

PR #1252 (2022-05-26): Fix EspressoDeleteTrackTest.
Fixes #817.
commit fb00f697 (2022-05-26): Fix EspressoDeleteTrackTest.

Fixes #817.
closed 2022-05-26 by dennisguse