OpenTracksApp/OpenTracks #820: Customizable Recording Layout

rgmf, 2021-06-16:
**Description**
We have a grid to show data fields in the recording layout:

[image]

From settings, users will been able to select the data fields they want to see and the order in the recording layout grid.

To do that we can use a “Preference action” that start a new Activity. (https://developer.android.com/guide/topics/ui/settings/customize-your-settings#actions). This Activity will use a RecyclerView with the drag and drop feature to offer to the user an UI to set their preferences (what to see and the order in the grid the fields data will be seen).

[image]

**Preferences Changes**
For every data field we save two pairs of key-value in shared preferences:
    • One of them to save the position in the grid.
    • The other one to set (true or false) the visibility.

So, we’ll define all these preferences keys in the values/settings.xml file:

stats_total_time_data_position_key
stats_total_time_data_show_key

stats_distance_data_position_key
stats_distance_data_show_key

stats_moving_total_time_position_key
stats_moving_total_time_show_key

stats_speed_position_key
stats_speed_position_show

stats_avg_speed_position_key
stats_avg_speed_show_key

stats_avg_moving_speed_position_key
stats_avg_moving_speed_show_key

stats_max_speed_position_key
stats_max_speed_show_key

stats_altitude_position_key
stats_altitude_show_key

stats_gain_position_key
stats_gain_show_key

stats_loss_position_key
stats_loss_show_key

stats_sensors_position_key
stats_sensors_show_key

Sensors clarification: sensors will add all user sensors connected. For simplification we offer to the user to show all or none of sensors.

**SettingsActivity Changes**
Add an option that will open a new activity ( https://developer.android.com/guide/topics/ui/settings/customize-your-settings#actions)

**StatisticsRecordingFragment Changes**
The layout will be a RecyclerView with a grid (GridLayoutManager). It’ll be like current stats sensors recycler view.

This new RecyclerView will get preferences information and put all stats where they should be according to the user preferences.

**Issues related**
https://github.com/OpenTracksApp/OpenTracks/issues/659
https://github.com/OpenTracksApp/OpenTracks/issues/302

**Discussion**
https://github.com/OpenTracksApp/OpenTracks/discussions/813

edern-appere, 2021-06-16:
This enhancement would be realy awesome. Many bike computers allow this kind of customization. 

One thing that could be great also is to allow to put some of the information displayed bigger than others. For example : when riding a bike, the current speed needs to be displayed in bigger size than average speed.

rgmf, 2021-06-16:
> One thing that could be great also is to allow to put some of the information displayed bigger than others. For example : when riding a bike, the current speed needs to be displayed in bigger size than average speed.

I'll consider it. For now, I have no idea how to do that un terms of UX/UI... Some idea?

edern-appere, 2021-06-16:
> > One thing that could be great also is to allow to put some of the information displayed bigger than others. For example : when riding a bike, the current speed needs to be displayed in bigger size than average speed.
> 
> I'll consider it. For now, I have no idea how to do that in terms of UX/UI... Some idea?

Well... when I wrote this, I had not read what you mentionned in the discussion #813 
At first my only goal was that it would be great to have the current speed at the same size than the informations displayed on the upper row (which currently are distance, activity and time). But I think your customization system would be better because it allows a full range of different settings for each type of rider and type of activity (and also depending on the sensors you have). Kind of like your Android homescreen : users try to arrange it so that it features the most commonly used apps.

dennisguse, 2021-06-17:
I like the idea for the configuration dialog!

Technical detail: I would rather store the UI information into one preference field (e.g., `speedCurrent;distance;speedAverage;elevation;gain;loss`).
It would be easier too handle (and to reset if we upgrade), way easier to figure out how it should look like (if one has to debug something), and we might need several layouts (by activity) in the future.
Potential option for sizes: `speed++;distance+;elevation;gain++`.

(future work) We could even add "empty" boxes (that show nothing), if somebody wants to align the items in an interesting manner ;)

A completely different alternative would be define separate layouts in XML and let the user choose one.
On the one hand, we might then end up with a lot of different layouts (combinations of show data, sizes etc) and might make our life miserable.
On the other hand, we could then also to weird layouts (e.g., first row only contains one element like speed; next row three etc).

Anyhow, it can only get better :partying_face:

rgmf, 2021-06-19:
> Technical detail: I would rather store the UI information into one preference field (e.g., `speedCurrent;distance;speedAverage;elevation;gain;loss`).
> It would be easier too handle (and to reset if [...]

Big technical detail!!! Thank you, I didn't found out such a possibility.

What I did at this moment is this:

```xml
<string name="stats_custom_layout_order_key" translatable="false">statsCustomLayoutOrderKey</string>
<string-array name="stats_custom_layout_order_values">
    <item>@string/stats_total_time</item>
    <item>@string/stats_moving_time</item>
    <item>@string/stats_distance</item>
    <item>@string/stats_speed</item>
    <item>@string/stats_average_moving_speed</item>
    <item>@string/stats_average_speed</item>
    <item>@string/stats_max_speed</item>
    <item>@string/stats_pace</item>
    <item>@string/stats_average_moving_pace</item>
    <item>@string/stats_average_pace</item>
    <item>@string/stats_fastest_pace</item>
    <item>@string/stats_altitude</item>
    <item>@string/stats_gain</item>
    <item>@string/stats_loss</item>
    <item>@string/stats_latitude</item>
    <item>@string/stats_longitude</item>
    <item>@string/stats_sensors</item>
</string-array>
```

Because we need translation strings. Later in the code I transform the array with default values in the string you pointed out: `speedCurrent;distance;speedAverage;elevation;gain;loss`

```java
public static List<String> getCustomLayoutOrder(SharedPreferences sharedPreferences, Context context) {
    final String defaultOrder = Arrays.asList(context.getResources().getStringArray(R.array.stats_custom_layout_order_values)).stream().collect(Collectors.joining(";"));
    final String value = getString(sharedPreferences, context, R.string.stats_custom_layout_order_key, defaultOrder);
    return new ArrayList<>(Arrays.asList(value.split(";")));
}

public static void setCustomLayoutOrder(SharedPreferences sharedPreferences, Context context, List<String> orderItems) {
    final String prefString = orderItems.stream().collect(Collectors.joining(";"));
    setString(sharedPreferences, context, R.string.stats_custom_layout_order_key, prefString);
}
```

closed 2021-08-21 by rgmf

edern-appere, 2021-09-09:
I would like to send you a big thank you for your hard work (coding, reviewing the code, exchanging ideas and suggestions) during this summer to make it possible to customize the recording screen ! Congrats to @rgmf and @dennisguse !! 

I've had a great opportunity to try it out last weekend as I went for a two day ride in the Schwarzwald with students and colleagues from the university I'm working for. OpenTracks really turned out to be incredibly useful. It really worked like a charm. The app was running all day long, screen turned on since it does not drain the battery of my phone. 

Ever since I first downloaded the app, more than a year ago, I thought it had the potential to be a true bike computer replacement and today, to me, the app has reached this goal. 
So happy to have this attached to my handlebar every time I reach for my road bike !

commit 1aa31461 (2021-06-24): Recording stats layout customizable.
Users can select the order and visibility of all stats for recording layout from the Settings.
Fixes #659.
Fixes #302.
Fixes #820.
commit 951b6fc5 (2021-06-24): Recording stats layout customizable.
Users can select the order and visibility of all stats for recording layout from the Settings.
Fixes #659.
Fixes #302.
Fixes #820.
commit 6c76e456 (2021-06-25): Recording stats layout customizable.
Users can select the order and visibility of all stats for recording layout from the Settings.
Fixes #659.
Fixes #302.
Fixes #820.
commit b3b23040 (2021-07-12): Recording stats layout customizable.
Users can select the order and visibility of all stats for recording layout from the Settings.
Fixes #659.
Fixes #302.
Fixes #820.