OpenTracksApp/OpenTracks #824: Update name of the activity in GPX file

vsevjednom-cz, 2021-06-20:
GPX is exported when I stop the measurement - that's ok but: Is possible to update a name of the activity in the existed GPX file when I changed it later? Simple solution: delete original GPX file and export a new - only when original GPX file exists.

dennisguse, 2021-06-20:
That might be an easy option, but also make our life messy.
What you actually want to do is to implemented synchronization between different devices (aka mobile and a server or even mobile and mobile).
And that is messy as you have to figure which one was the most recent version.

Implementing an overwrite option would (in my opinion) require at least a check if the file that we are going to overwrite was not changed after it was exported.

An example for this is the use of [Syncthing](https://syncthing.net/) to send files to another device and both might change the files.

Related: https://github.com/OpenTracksApp/OpenTracks/issues/90

vsevjednom-cz, 2021-06-20:
It looks like synchronization between devices but no. It's only synchronization in OpenTracks. I can synchronize it with other devices after that. I want new export (with new nameof activity) only if GPX file exists 🙃.

vsevjednom-cz, 2021-08-17:
I've got improvement of my idea. Update of the name of activity in GPX file sounds unnecessary. Better would be: export for each activity individually. Sometimes I need export of only one activity, not all. For example if here is some error (#907). Individual export is better and easier than resume and stop tracking.

dennisguse, 2021-08-23:
Could you use the share action for this?

vsevjednom-cz, 2021-08-23:
Yes, it works. But it's weird a little bit. Share to the email is ok but share via Messenger (FB) sends only `ic_logo_24dp` image without another attachment. Individual export would be cute improvement.

closed 2024-11-21 by dennisguse