OpenTracksApp/OpenTracks #825: Create a hidden file in the export directory

vsevjednom-cz, 2021-06-20:
I've got the Android 9.1.0. Export directory is set.  But Android deletes directory everyday when directory is empty and when I create new directory in root directory for GPX export.

So is possible to create "hidden" file in this directory?

Like this:
```
DocumentFile ... = PreferencesUtils.getDefaultExportDirectoryUri(sharedPreferences, getApplicationContext());

...

File filepath = new File(path + File.separator + ".folder");

File filepath = new File(path + File.separator + ".folder");
            if(!filepath.exists()){
                FileOutputStream fos = null;
                try {
                    fos = new FileOutputStream(filepath);
                    byte[] buffer = "".getBytes();
                    fos.write(buffer, 0, buffer.length);
                    fos.close();
                }
...
            }
```

dennisguse, 2021-06-20:
That is possible, but it would be better to figure out why an empty folder gets deleted (or rather by whom).
At least on my devices, I never saw such a behavior.

vsevjednom-cz, 2021-06-21:
Hmm, Optimizer deletes empty directories maybe. There is Settings -> Auto-clean junk files. Your app can limit this weird function.

dennisguse, 2021-06-21:
@vsevjednom-cz Can you check that this is not specific to your Android variant?
I have never seen such an option and I am reluctant to add workarounds for such things.
We have enough trouble with complexity already and adding more complexity is more like an anti-feature.

vsevjednom-cz, 2021-06-22:
You're right. It's weird function of my Android 9.1.0 EMUI. I enabled it earlier maybe.

closed 2021-06-22 by vsevjednom-cz