OpenTracksApp/OpenTracks #826: Recording Settings Definitions

SlidingHorn, 2021-06-20:
This is probably a stupid question, but I was curious as to what exactly the "Recording Distance Interval" and "Max Distance" settings are.

The former _sounds_ redundant (if "Recording Time Interval" is used), which makes me think it does something other than what I originally thought.

The other, I have no idea what it does.

The reason I'm curious is that I also use my Tracks as GPS Traces to upload and use for editing OpenStreetMaps, so in my particular case, knowing what exactly what the settings do and if they can be made more precise is important to me.

Thanks in advance for any clarification!

dennisguse, 2021-06-20:
* Recording time interval: how often do we want to get updates from the GPS?
* Max distance: create a new segment if the last location is farther away than XXX
* GPS accuracy: ignore every GPS measurement with an accuracy larger than XXX
* Recording distance interval: ignore every GPS measurement if the last location it is closer than XXX 

Does this help?

PS: We should probable re-translate all of those.

SlidingHorn, 2021-06-20:
> * Recording time interval: how often do we want to get updates from the GPS?
> * Max distance: create a new segment if the last location is farther away than XXX
> * GPS accuracy: ignore every GPS [...]

That helps immensely. Thank you for clearing that up for me.

dennisguse, 2021-06-20:
PS: If a sensor (like heart rate) is connected, we ignore the `Recording distance interval` and store the most recent measurement on each GPS measurement.

closed 2021-06-20 by dennisguse