OpenTracksApp/OpenTracks #828: distance, time, speed, and geolocation values flicker between two values very fast.

montagsmodell, 2021-06-20:
I am currently trying to add Moofit Cadence + Speed sensors to the app but it did not work so far. Currently, only the cadence value is displayed correctly. The speed value is way too low (like 3 km/h). Yesterday I saw that a new version of the app was available via f-droid. Version number is v3.18.4-9234da49.
After download I tried to pair both sensors again but it still doesn't work. Additionally, now I have rapidly changing values for the fields mentioned in the title.
Like "current speed": 25km/h and then 0km/h the next second. Geolocation is first correct and in the next moment only "-" is displayed. Maybe a problem with my GPS? Anyway, I tried to resolve it by restarting app and device but it did not change the outcome.

Maybe anyone has some input on either of the two issues I am facing with the app. Otherwise it is a real great app!

**Technical information**
 - Device: [Oneplus 6t]
 - OS: [LineageOS Fajita]
 - Version [v3.18.4]

sebastianha, 2021-06-20:
Same here, the latest version 3.18.4 behaves wird on the main screen while recording. 3.18.3 was fine.

rgmf, 2021-06-20:
@dennisguse we introduce this bug here: https://github.com/OpenTracksApp/OpenTracks/commit/2f0449c4209633417d798b7b0f108a99f46956ad

I remember I tested OT before last version 3.18.4 and I don't understand how didn't I stumble upon this bug?

@sebastianha @montagsmodell thank you so much for open/comment it.

montagsmodell, 2021-06-20:
happy to help! regarding the cadence + speed sensor issue: Do you know if it is possible to pair the two moofit sensors at once?

dennisguse, 2021-06-20:
Good news it only affects the UI while recording; the actual data is okay.

@rgmf you are right and we both overlooked it ;)
I introduced this issue while working on the speed sensor (and tested without using GPS at all).
The problem is that we sent different data to the UI when a GPS point is received and for the every 1s update.
This is already fixed (but not released): https://github.com/OpenTracksApp/OpenTracks/commit/b947a727e7963e40ac69e62c221e3f190b6ca5b3
I just wanted to do another test drive.

Moreover, I just realized that we don't need to update the UI when a GPS point is received.
The 1s update should be sufficiently doing the job (we need this anyhow to update the time).

@montagsmodell @sebastianha Thanks for reporting it so quickly!

@montagsmodell regarding the sensors: should be working; if doesn't please open a new issue.
v3.18.3 the sensor parsing is broken and v3.18.4 the UI is broken (but sensor parsing is working).
So, collecting data "should" be working.

commit 2d60b12b (2021-06-21): Bugfix: TrackRecordingService now computes EGM2008 for UI (i.e.,  StatisticsRecordingFragment).

Fixes #828.
PR #831 (2021-06-24): Bugfix: TrackRecordingService needs to send data from lastTrackPoint …
…in it's 1s updates.

This also removes UI updates if a GPS measurement is received as 1s update should be sufficient.

Fixes #828.

BUT: elevation is shown in WGS84 as the EGM2008 transformation is happening in the TrackDataHub.
TrackRecordingService needs to do the same.
commit 3030b4df (2021-06-24): Bugfix: TrackRecordingService needs to send data from lastTrackPoint in it's 1s updates.

This also removes UI updates if a GPS measurement is received as 1s update should be sufficient.

Fixes #828.
commit b5555e30 (2021-06-24): Bugfix: TrackRecordingService now computes EGM2008 for UI (i.e.,  StatisticsRecordingFragment).

Fixes #828.
closed 2021-06-24 by dennisguse