OpenTracksApp/OpenTracks #833: Cycling speed and distance sensor: returned weird value (way too large)

dennisguse, 2021-06-21:
Looks like an overflow issue.
Problematic data: 
```
<trkpt lat="XX" lon="XX">
<ele>72</ele>
<time>2021-06-21T17:21:23.966Z</time>
<extensions><gpxtpx:TrackPointExtension>
<gpxtpx:speed>8.3</gpxtpx:speed>
<opentracks:gain>0</opentracks:gain>
<opentracks:loss>0</opentracks:loss>
<opentracks:distance>0</opentracks:distance>
</gpxtpx:TrackPointExtension></extensions>
</trkpt>
<trkpt lat="XX" lon="XX">
<ele>67</ele>
<time>2021-06-21T17:21:33.954Z</time>
<extensions><gpxtpx:TrackPointExtension>
<gpxtpx:speed>577947520</gpxtpx:speed>
<opentracks:gain>0</opentracks:gain>
<opentracks:loss>0</opentracks:loss>
<opentracks:distance>9,169,715,200</opentracks:distance>
</gpxtpx:TrackPointExtension></extensions>
</trkpt>
```

Device: Garmin Sensor Speed 2 (software revision unknown).

dennisguse, 2021-06-21:
Hypotheses:
* (A) Overflow issue (distance from sensor).
* (B) Sensor provided broken data (distance from sensor)
* (C) Distance was computed by GPS
  => previous TrackPoints were recorded every 1s; but this one was took 10s

For (A)
```
                long wheelDiff = UintUtils.diff(wheelRevolutionsCount, previous.wheelRevolutionsCount, UintUtils.UINT32_MAX);
```

Just checked the sensor data (raw rotation count):
2021-06-13: 381,616
2021-06-21: 457,371

For the time interval the following would be expected: 10s at ca. 30km/h => ca. 8m/s => 80m-100m => ca. 50 wheel rotations

dennisguse, 2021-08-23:
Did not happen again.
Closing for now.

rgmf, 2021-09-12:
I get 4.5374991E9 value as a cadence while working on #931.

I can reproduce it and I did it with main branch.

To reproduce (I did it several times):
- Start a recording with a cadence sensor.
- Stop pedaling for more than 20 minutes (I don't know where threshold is).
- Restart pedaling and you'll see a big value for cadence.

I can see the following values for `crankRevolutionsCount` from the beginning of the recording:
```
- Computation 1:
current crankRevolutionsCount = 2
previous crankRevolutionsCount = 0
new cadence value = 25.377943

- Computation 2:
current crankRevolutionsCount = 3
previous crankRevolutionsCount = 2
new cadence value = 38.76341

- Computation 3:
current crankRevolutionsCount = 4
previous crankRevolutionsCount = 3
new cadence value = 40.156864

- Computation 4:
current crankRevolutionsCount = 5
previous crankRevolutionsCount = 4
new cadence value = 38.568737

- Computation 5:
current crankRevolutionsCount = 6
previous crankRevolutionsCount = 5
new cadence value = 22.053123

---- Here I stop pedaling for more than 20 minutes -----

- Computation 6:
current crankRevolutionsCount = 2
previous crankRevolutionsCount = 6
new cadence value = 4.5374991E9

- Computation 7:
current crankRevolutionsCount = 3
previous crankRevolutionsCount = 2
new cadence value = 38.424015
```

Look at `Computation 6`. There you have an overflow that comes from here:
`UintUtils.diff(crankRevolutionsCount, previous.crankRevolutionsCount, UintUtils.UINT32_MAX);`

`crankRevolutionsCount` cannot be less than `previoud.crankRevolutionsCount`.

I'm going to try to fix it on #931 but I re-open it if you don't mind @dennisguse

dennisguse, 2021-09-12:
Amazing work!
I did not know that the counter reset.

closed 2021-09-21 by rgmf

dennisguse, 2021-09-22:
@rgmf do you think, we should create another release for these changes? No new feature, but a lot of bug fixes :)

rgmf, 2021-09-22:
> @rgmf do you think, we should create another release for these changes? No new feature, but a lot of bug fixes :)

Yes, I'd do it. Bug fixes are always welcome :D So, go ahead.

dennisguse, 2021-09-22:
@rgmf Do you think, we should also include #943 in the upcoming (bugfix) release? From my side: looks okay, and we can do some fine tuning later.
In any case, it is just a settings dialog :D

PS: I have only a phone at hand, so you would need to create the release ;)

timoschwarzer, 2021-09-22:
@dennisguse I'll not be able to update the PR for at least two weeks, so yeah go for it :)

rgmf, 2021-09-22:
@dennisguse Yes, I'd also include #943. Like you say we can improve it later.

So, I'm going to rebase and merge #943 and I create the release ;)

commit d9bf7f75 (2021-09-18): Cadence set to 0 when connected but user is not pedaling. Fixes #931.
Speed set to 0 when connected but last value is outdated. Fixes #930.
Bug: overflow when reset sensors (cadence and speed). Fixes #833.
Moving time stops when is not moving (the last track point is not moving) and restarts when last track point is moving.
commit d99fb5ea (2021-09-21): Cadence set to 0 when connected but user is not pedaling. Fixes #931.
Speed set to 0 when connected but last value is outdated. Fixes #930.
Bug: overflow when reset sensors (cadence and speed). Fixes #833.
Moving time stops when is not moving (the last track point is not moving) and restarts when last track point is moving.