OpenTracksApp/OpenTracks #836: Avg moving speed is over/under Avg speed

vsevjednom-cz, 2021-06-22:
**Describe the bug**
I'm a little confused.
"Avg speed" is over "Avg moving speed" when GPS tracking is active.
But: "Avg speed" is under "Avg moving speed"  when I view older activity.

Is possible to unify it? For example "Avg moving speed" over "Avg speed".

rgmf, 2021-06-23:
I'm working on #820 and I think #820 would fix this one, right?

vsevjednom-cz, 2021-06-23:
Yes, it looks like fix. Will be the same layout for tracking and not tracking..?

rgmf, 2021-06-23:
> Will be the same layout for tracking and not tracking..?

They cannot be the same because Current Speed wouldn't make sense for recorded tracks, for example.

vsevjednom-cz, 2021-06-23:
I mean only the same layout of average speed 🙃.