OpenTracksApp/OpenTracks #837: Should OT work with a Garmin Vivosmart 4?

SlidingHorn, 2021-06-23:
I'm looking into getting a wearable, and the Vivosmart 4 is standing out to me. I'm wondering if you knew/had any confidence as to whether it would work with OpenTracks in terms of sharing heart rate and improving GPS?

dennisguse, 2021-06-23:
Nope and the device is not supported by Gadgetbridge.

Regarding HR: we support everything that provides the BLE profile for continuous measurement.
Some of the smartwatches like Miband can do this.
I can recommend to getting in contact with the Gadgetbridge team.
Smartwatches is (sadly) a hell of proprietary solutions.

Regarding GPS (for recording): not so far.
We can only use the internal GPS.
And I don't know if any of the Garmin smartwatches can actually share this.

Regarding GPS precision: depends on how old your phone is.
My good old Samsung S4 way less precise than my Pixel1 (and way slower).

PS: What is the actual problem you have?
You have a Vivosmart4 and want to use it while running to get HR + live information?
Or why do you need a smartwatch?

dennisguse, 2021-06-23:
And the Vivosmart4 only supports ANT+ for continuous HR.
https://de.manuals.plus/garmin/vivosmart-4-Handbuch/amp#bluetooth_settings

And that would require adding closed source libraries and is therefore not an option.
So, that is not going to work...

SlidingHorn, 2021-06-24:
> PS: What is the actual problem you have?
> You have a Vivosmart4 and want to use it while running to get HR + live information?
> Or why do you need a smartwatch?

I'm looking for a wearable that allows me to track my rides and my overall fitness. I've looked at Gadgetbridge to see what would be compatible, so I've refocused my search toward those devices.

I guess my remaining question would be: If I end up getting a watch with GPS, would OpenTracks use that as well as the GPS on my phone or is it one or the other (and which of the two would likely be more accurate? Again, I use the GPX tracks for OpenStreetMap editing as well, so being more accurate is important, and some of the time, the accuracy on my phone is a little less accurate than I'd like.)

dennisguse, 2021-06-24:
No.

TL;DR
Not directly. You may use the smartwatch to record your tracks (if it has a GPS) and then export the data via Gadgetbridge (if supported). Then you can import these tracks into OpenTracks, but I don't see the benefit.
OpenTracks only supports internal GPS.

I am not even sure if there are smartwatches out there that support this in some standardized (aka non proprietary way.

closed 2021-06-24 by dennisguse