OpenTracksApp/OpenTracks #839: testing and stable releases

nutpantz, 2021-06-23:
I love open tracks, but some times serious bugs happen which can really ruin a ride, run etc
Please b consider a stable branch and a testing branch so those who really need it to work have the fewest issues while giving those who want the bleeding edge the ability to test and report bugs the ability to do so.

 your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

dennisguse, 2021-06-23:
Suggestions and resources are welcome.

TL;DR 

I couldn't agree more.
We had some stability incidents so far and, you are right, most recently some ugly ones were introduced as we started working on ancient parts (and most of them were introduced by @dennisguse).

If you have resource to spare, feel free to contribute by:
* manual testing (individual PRs and before releases)
* provide nightly builds: https://github.com/OpenTracksApp/OpenTracks/issues/416
* add tests (e.g., https://github.com/OpenTracksApp/OpenTracks/issues/778, https://github.com/OpenTracksApp/OpenTracks/issues/574)

For me, OT is a spare time (aka hobby) project that I use (that's why we have BLE sensors), where I learn stuff (that's why we have EGM2008 support) and fun with others (to name some: https://github.com/OpenTracksApp/OpenTracks/graphs/contributors).

@nutpantz If you want to work on stability that would be pretty awesome!

dennisguse, 2021-07-29:
Superseded by #894.

closed 2021-07-29 by dennisguse