OpenTracksApp/OpenTracks #840: OpenTracks vibrates excessively when trying to pause or stop

gdt, 2021-06-23:
OpenTracks causes vibration way too much, and it's hard to pause and stop.
I am up to date via f-droid, 3.18.4.

To reproduce,  start a track, and then hit pause or stop.  Notice that it buzzes once, leading you to think it did what you asked.  Then realize it didn't, and push/hold and endure a long period of buzzing before it takes.

I realize there is a need to avoid falsing, but it seems far better to have a scheme that doesn't have this long-buzz behavior.   Perhaps just a lock/unlock state for the UI where you have to touch twice different places to unlock and then things just work, and auto-lock on track start, or make it settings.

I also realize that the behavior is likely intended, but as a user with no way to turn it off, it really feels like a bug.

dennisguse, 2021-06-23:
Yes, that was intended.
We implemented this in https://github.com/OpenTracksApp/OpenTracks/issues/241

What would be your actual suggestion?

PS: PR welcome

dennisguse, 2021-06-23:
@gdt Please be careful with labeling issues with `(potential) bug`.
This indicates something is really broken and usually kind of urgent.

gdt, 2021-06-23:
As a user, this made me almost think about uninstalling.  But I see your point that this is preference.

Reading over that linked issue, I think what's really wrong in the current scheme is abuse  of vibration.   In my view vibration should only be used as a component of a notification, or very very brief as a confirmation tone.

So my two concrete suggestions are:

  * Add a preference for "use vibration on pause/stop" that I can turn to NO
  * Stop using vibrate at all, and when the pause/stop is pushed throw some popup that says "Hold for 2s to pause" (stop) to convey this surprising situation to the user who has no idea to expect this.
  * Maybe also had a "hold 2s" notation above pause/stop **all the time** so that people will know they will have to do that (and again don't vibrate).

Part of where I'm coming from is that vibration stuck on leads me to think "my phone is broken", not "somebody thinks this is a good idea".

dennisguse, 2021-06-28:
Let's see what we can do (aka other stuff is more important).

So, far we display a tiny message informing the users that they need to keep the button pressed (and we vibrate).

underr-ua, 2021-07-05:
When i start recording a track and then short press on stop recording (something in the middle between really short press and hold the button, recording doesn't stop) i get constant vibration. I see such behavior  in all latest versions.

dennisguse, 2021-07-05:
@underr-ua That sounds like another issue can you a create a new one for this (incl. some more details like phone etc)?

underr-ua, 2021-07-05:
Ok, I'll create another ticket.

closed 2024-12-19 by dennisguse