OpenTracksApp/OpenTracks #841: Inaccuracy in Czech translation

vsevjednom-cz, 2021-06-24:
**Describe the bug**
`Obnovit skladbu` in czech translation means something like "restore song". Czech language is one of the hardest languages at the world 😋.
Look at the Google translator: https://translate.google.cz/?sl=cs&tl=en&text=Obnovit%20skladbu

**To Reproduce**
Tap to menu 📄.

**Technical information**
`<string name="menu_resume_track">Obnovit skladbu</string>` in `cs/strings.xml`

rgmf, 2021-06-24:
@vsevjednom-cz thank you for open it.

If you want you can fix it and open a PR. Although the best for translations is enter here and write/change the strings: https://weblate.bubu1.eu/engage/OpenTracks/

It would be great ;)

vsevjednom-cz, 2021-06-24:
I tried to edit translation of `Resume track`. I hope you see it. I'll fix other inaccuracies of translations in future 🙃.

dennisguse, 2021-06-25:
@vsevjednom-cz I just checked and I saw the translation - not yet in OpenTracks as Weblate has a grace period for translations (to not commit every single change).
Before every release (at least) I pull the localization updates.

closed 2021-06-25 by dennisguse