OpenTracksApp/OpenTracks #843: Shared Track - vertex height attribute

vkbellis, 2021-06-24:
**Is your feature request related to a problem? Please describe.**
No, not exactly.

**Describe the solution you'd like**
I'm just curious if the exported line feature from OpenTracks (KMZ format) and in particular, the vertices, if the Elevation attribute is supposed to be ellipsoid height?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

dennisguse, 2021-06-24:
Elevation is in  Height Above Ellipsoid (i.e., WGS84) for all exported formats.
For the UI, we compute EGM2008.

closed 2021-06-24 by dennisguse

vkbellis, 2021-06-24:
What is the best way for the user to capture ortho heights (H) from OpenTracks?

dennisguse, 2021-06-24:
Don't know what that means.
Do you mean height above ground?

vkbellis, 2021-06-24:
EGM2008 is the gravity model (geoidal separation) that you mentioned using which allows orthometric heights to be calculated. Are those heights able to be saved or captured in a file?

dennisguse, 2021-06-24:
That would be possible, but you can also add this info in post-processing.
For the UI, we are using the data from GeographicLib (and hopefully the same implementation): https://geographiclib.sourceforge.io/html/geoid.html

Adding export is not too complicated, but it adds additional complexity.

What is the problem you want to solve?

vkbellis, 2021-06-24:
_Would be_ sounds like it isn't currently, and that's fine, I was just asking.

The reason for my asking is because the Samsung Galaxy SM-A716U has erroneous geoidal separation values. Somebody mentioned that OpenTracks uses the geoid model EGM2008 and I wondered if an exported file would support that claim.

dennisguse, 2021-06-24:
Now I would be interested in a link :)

OpenTracks just uses Android's LocationManager and that should provide altitude in WGS84.
Whatever the actual hardware/driver does is beyond our control.
We are also not parsing the NMEA response (as GPSTest does).

And OpenTracks stores and exports, what it gets from the LocationManager (we assume WGS84) and then we correct this value using EGM2008 correction data to get a better height above sea level.

vkbellis, 2021-06-25:
I'm unsure which link you might be referring to, but maybe this was what you wanted(?)
https://github.com/barbeau/gpstest/issues/530

The exported KMZ contains ellipsoid heights (h). Consider option to export orthometric heights (H); e.g., EGM2008.

Red dots are OpenTracks ellipsoid heights. Green is terrain model from lidar.
[image]

dennisguse, 2021-06-25:
@vkbellis Now I am confused :)

1. Do you have a link for "Samsung Galaxy SM-A716U has erroneous geoidal separation values."?
2. We are exporting to GPX WGS84 as defined in the standard:  `Height (in meters) of geoid (mean sea level) above WGS84 earth ellipsoid. As defined in NMEA GGA message. ` (https://www.topografix.com/gpx/1/1/gpx.xsd)
3. We are exporting to KML/KMZ WGS84 aka `h`. But I do not find a definite documentation if this is okay. I saw some reference saying that it should be EGM96. `The vertical component (altitude) is measured in meters from the WGS84 EGM96 Geoid vertical datum.` (https://en.wikipedia.org/wiki/Keyhole_Markup_Language); this seems to be NOT specified in the standard for KML2.2/KML2.3.

@vkbellis Can you take the endeavor to figure if we can export EGM2008 to KML's `<altitude>` that would be great.
Changing this takes me 15min.

Thanks for the link to https://github.com/barbeau/gpstest/issues/530.
I did not know this, but happy if they take the code :)

vkbellis, 2021-06-25:
Samsung's erroneous value is given in the NMEA sentences, for example:
GPSTest v3.9.12 (18089-google), Manufacturer: samsung, Model: SM-A716U, GNSS HW Year: 2019, Platform: 11, API Level: 30 
`$GNGGA,215300.00,4432.673982,N,06825.027815,W,1,12,0.6,66.7,M,-45.0,M,,*7B`

44.5445663667, -68.41713025 
[GeoidEval](https://geographiclib.sourceforge.io/cgi-bin/GeoidEval?input=44.5445663667+-68.41713025+&option=Submit): -25.3123 m

The severity of the error in the geoidal separation value appears to be dependent on the device, for example:
GPSTest v3.9.12 (18089-google), Manufacturer: samsung, Model: SM-T970, GNSS HW Year: 2016, Platform: 10, API Level: 29 
`$GPGGA,174820.96,4432.667082,N,06825.033167,W,1,10,1.6,56.5,M,-29.1,M,,*6B`

GPSTest v3.9.12 (18089-google), Manufacturer: motorola, Model: Moto E (4) Plus, GNSS HW Year: 2016, Platform: 7.1.1, API Level: 25
`$GPGGA,181709.011,4432.6833,N,06825.0227,W,0,0,,36.3,M,-30.4,M,,*4A`

Thank you for the clarification in regards to the GPX. I'll try that immediately :) ...
uh, looks like GPX heights are also ellipsoidal
[image]

Sorry, I have no programming experience.

dennisguse, 2021-06-25:
Now I understood it ;)

@vkbellis this is not a programming issue - more an information retrieval problem.
I did not find any definitive answer for KML/KMZ to which reference model the altitude should be encoded.
I just have limited to time read through all these standard documents.

So (for now), OpenTracks export WGS84 HAE for both formats.
For KML/KMZ, this might not be correct and we may be able to use EGM2008.
But we have to be careful for interoperability with other applications (also regarding OpenTracks import functionality).
That's why I am reluctant to change it.