OpenTracksApp/OpenTracks #844: Altitude gain and loss not available

smu55, 2021-06-25:
Hi
As you can see in the screenshots the gain and loss of altitude/elevation are not shown although the actual altitude is measured, neither live nor in the stats. So it must be simply a calculation issue. I use a Samsung Galaxy a51 with German translation.
Thanks for looking into it.

[image]
[image]
[image]

rgmf, 2021-06-25:
I suppose your smartphone doesn't have barometric sensor, right? Maybe this can help you: https://github.com/OpenTracksApp/OpenTracks/issues/756

dennisguse, 2021-06-25:
@rgmf I checked: the a50 does not have a barometric sensor and thus no data is shown.
Deriving gain/loss from GPS (kind of noisy elevation data) is not easy as summing a noisy signal is a bad idea.

* Workaround aka hide the missing data: https://github.com/OpenTracksApp/OpenTracks/issues/791
* Actual fix (do some magic math): https://github.com/OpenTracksApp/OpenTracks/issues/619

PS: The German translations are kind of hard on the layout.
I guess, I should use OpenTracks in German for some weeks :|

closed 2021-06-25 by dennisguse

rgmf, 2021-06-25:
> @rgmf I checked: the a50 does not have a barometric sensor and thus no data is shown.
> Deriving gain/loss from GPS (kind of noisy elevation data) is not easy as summing a noisy signal is a bad idea.
> 
>     * Workaround aka hide the missing data: [Workaround for #619: no barometric sensor => no gain/loss in IntervalStatistics #791](https://github.com/OpenTracksApp/OpenTracks/issues/791)
> 
>     * Actual fix (do some magic math): [elevation gain/loss without barometer #619](https://github.com/OpenTracksApp/OpenTracks/issues/619)

I opened a PR with the workaround.

> 
> PS: The German translations are kind of hard on the layout.
> I guess, I should use OpenTracks in German for some weeks :|

A word with 17 letters and only 4 vowels!!!! It blows up spanish people mind :rofl: :rofl: