OpenTracksApp/OpenTracks #845: Gear calculation

smu55, 2021-06-25:
Hi,
If I have a cadence and speed sensor, knowing the wheel circumference (ex 622/25) and the tooth numbers of my cassette (ex 12-25) and chainring (ex 50/35, all manually inserted data), it should be possible to live calculate the actual gear I am using in every situation. This data could also help evaluate in post the usage of gears. Theoretically one could even calculate the actual estimated power applied, considering the weight and elevation gain, lacking only the influence of the wind (which could be estimated as well).
That would be a useful and unique function for a tracking app.

dennisguse, 2021-06-25:
In theory yes: but why?
And you could do this in post-processing if you wanted to.

TL;DR
And the cadence does not tell you how much power was actually applied and thus computing some kind of wear gets difficult.
And the actual wear is created by the stretching the chain over time, so it affects the wear of all gears.
And changing individual parts of the cassette is (in my price range) actually not possible.

@smu55 That doesn't mean it cannot be built or might not be useful, but I would prefer at least some proof of concept (datawise).

PS: Estimating power is not worth the effort as it will be really noisy.
Estimating elevation gain is hard as well as the impact of weight.
Better to buy a power meter (and even those give you 2% accuracy; and need calibration).

dennisguse, 2021-07-29:
@smu55 would you invest time to build this?

dennisguse, 2021-07-29:
@smu55 Please re-open if you want to work on this.

closed 2021-07-29 by dennisguse