OpenTracksApp/OpenTracks #853: Display while recording doubles values, but flips momentarily to correct ones

ytc-1, 2021-06-28:
**Describe the bug**
When recording trip date the display is showing double everything (time, average speed, moving speed etc).
If I watch it while cycling (not safe :-)) I note that every 15 seconds it flips back to the true readings.

When the recording has finished, the actual/true readings are given.

**To Reproduce**
Start recording
Start moving
Get confused by output telling you you have ridden 2 miles when it is 1 mile :-)

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: Pixel 4a
 - OS: Android 11
 - Versionkernel 4.14.212

dennisguse, 2021-06-28:
Duplicate of #828 and fixed in v3.18.5 (not yet on F-Droid).

TL;DR
Was a fuck up on my side while testing only with a cycling sensor indoors.
Sorry for the inconvenience and thanks for reporting it!

The problem now is that v3.18.5 is not yet build/distributed via F-Droid.
... and they seem to have some build issues again - last run on there side was on the 25th of June.
So, it may take at least 2-3 days before the v3.18.5 becomes available.

You may check this page to see the build status for F-Droid: https://f-droid.org/wiki/index.php?title=Special:RecentChanges&days=7&from=&hidebots=0&hideanons=1&hideliu=1&limit=500

or use @pstorch's Build Status app for F-Droid: https://f-droid.org/en/packages/de.storchp.fdroidbuildstatus/

closed 2021-06-28 by dennisguse

ytc-1, 2021-06-29:
>
> Duplicate of #828 
> <https://github.com/OpenTracksApp/OpenTracks/issues/828> and fixed in 
> v3.18.5 (not yet on F-Droid).

Sorry, I'd read the bug list and and didn't see this as a duplicate. 
Maybe I should have gone to the forum chat 1st?

> TL;DR
> Was a fuck up on my side while testing only with a cycling sensor indoors.
> Sorry for the inconvenience and thanks for reporting it!
>
> The problem now is that v3.18.5 is not yet [...]

dennisguse, 2021-06-29:
@ytc-1 Don't worry; I forgot (again) about delay for F-Droid - I should have pinned the issue, so people can find it easily.
Creating an issue was fine as it is a real problem.
Github's discussion is more for QA question, but we have not really used it that much - you can also create an issue for this :)

Have a nice day!