OpenTracksApp/OpenTracks #854: Version 3.18.5 Still Not on F-Droid

SlidingHorn, 2021-06-30:
I've been anxiously waiting for 3.18.5 to come across on F-Droid due to issue #828, and it still hasn't come across.  I know it usually takes a couple days for an update to be found & built, but I feel like we're outside the normal time window.

Was there something weird with the latest release?

SlidingHorn, 2021-06-30:
Okay, so I bugged the nice people in the F-Droid matrix chat room and had someone (slowly) explain to me how the build process works, and it turns out, the previous build (which kicked off just a few hours before 3.18.5 was released) had been in limbo until  awaiting signing and publishing until today.

That kicked off another build cycle today, in which OpenTracks was included.

closed 2021-06-30 by SlidingHorn

dennisguse, 2021-06-30:
Sometimes it (sadly) just takes time.
Thanks for taking care!

... and usually I am not careful enough when drafting releases.
"Works for me"-style...

dennisguse, 2021-07-02:
v3.19.5 is now also on F-Droid :partying_face:

SlidingHorn, 2021-07-02:
Grabbed it this morning!  I'd been refreshing the buid page regularly lol

dennisguse, 2021-07-02:
I know this feeling :)