OpenTracksApp/OpenTracks #860: Replace UUID with ULID

matrixik, 2021-07-01:
**Is your feature request related to a problem? Please describe.**
Replace UUID with [ULID](https://github.com/ulid/spec) so exported names can be sorted lexicographically.

**Describe the solution you'd like**
It would fix people complains about wrong order of exported files (like https://github.com/OpenTracksApp/OpenTracks/issues/512#issuecomment-817359415) and still provide unique identifier for tracks.

**Describe alternatives you've considered**
[KSUID](https://github.com/search?l=Java&q=ksuid&type=Repositories) (more info: https://segment.com/blog/a-brief-history-of-the-uuid/)

**Additional context**
More explanation: https://www.honeybadger.io/blog/uuids-and-ulids/

dennisguse, 2021-07-01:
If those are binary compatible and solve a problem - why not.

Some things to consider:
* Gadgetbridge also needs to be changed as they nowadays generate `opentracks:uuid` as well.
* `opentracks:uuid` in the export/imports is required to be backward compatible

And I am (personally) not a fan of adding external dependencies to OpenTracks as it complicates long-term maintenance.
So, far we managed to just use Androidx.

vanous, 2021-07-11:
Let us know how you decide to proceed @dennisguse , so we can follow as necessary.

dennisguse, 2021-07-11:
@vanous Not my problem - if somebody wants it, they have to built it.
Being backward compatible will be kind of painful though.
:man_shrugging: 

@vanous Thanks ;)

dennisguse, 2022-01-09:
Does not seem that important.

closed 2022-01-09 by dennisguse