OpenTracksApp/OpenTracks #861: Question: where/how can I get activity ID?

vsevjednom-cz, 2021-07-01:
I want to update activity name/desc (`TrackUtils.updateTrack`) in my own `AbstractActivity`. But I see only `opentracks:trackid` in XML (`e03aa77a-a7d6-40ad-9fe2-af436247d0ef` for example). But `TrackEditActivity` uses some id of activity (258 for example) instead of `opentracks:trackid`.

Where/how can I get this activity id?

dennisguse, 2021-07-01:
What you are looking for is `Track.getId()` vs. `Track.getUuid()`.
And you may take a look at: `ContentProviderUtils.getTrack(uuid vs. TrackId)`

Internally, we always use `Track.getId()` and for import `Track.getUuid()` as `Track.getId()` is specific to the database.

vsevjednom-cz, 2021-07-01:
Thank you, nice! But how to call `ContentProviderUtils.getTrack`? I know UUID from the GPX.

I tried something like this:
```
Track.Id trackId = getIntent().getParcelableExtra(SOME_UUID); or UUID trackId = getIntent().getParcelableExtra(SOME_UUID);
contentProviderUtils.getTrack(trackId);
```
But there's `FATAL EXCEPTION`:
`java.lang.NullPointerException: Attempt to invoke virtual method '...' on a null object reference}`

😥 😓

dennisguse, 2021-07-02:
For trying it out: just hardcode the UUID.

vsevjednom-cz, 2021-07-03:
🤪 But where? I'm trying it again and again without success 😟. Or do you mean hardcode in `TrackEditActivity`? I don't want to edit code of OpenTracks... so... my problem is solved. I'll not to do anything 😋.

dennisguse, 2021-07-03:
Yes.

Posting a stacktrace usually helps + I would rather move this into the discussion section.

dennisguse, 2021-07-03:
Here: https://github.com/OpenTracksApp/OpenTracks/discussions

PS: and do you have your code in branch?
I am not sure where you get the NPE.

Regarding stacktrace: usually, the last three lines are the most helpful (message + 2x `at`)

vsevjednom-cz, 2021-07-03:
I don't want to edit OpenTracks code - how you advised me earlier. So another idea: is possible to add id of activity into GPX export? Into tag named `opentracks:activityid` for example.

dennisguse, 2021-07-03:
What do you mean by activityId?
* the internal database id aka `Track.Id`, 
* the exported `opentracks:trackid` aka UUID, or
* some id for the so-called category (e.g., cycling, hiking etc), or
* something else?

And what is your use-case?

vsevjednom-cz, 2021-07-03:
I mean `Track.Id`. I'll save this `Track.id` into my MySQL and later I can update the name and description of the activity in my `AbstractActivity`.

dennisguse, 2021-07-04:
Track.Id: is local to the device's sqlite database
Track.getUuid() / opentracks:trackid: is (almost) guaranteed to be global unique

Internally, OpenTracks uses `Track.Id` (as it is faster to handle for the database), but we do lookups also via `uuid` (on import to prevent duplicated imports).

TL;DR

I would advise against storing the `Track.Id` into your MySQL.
This identifier is _local_ to the device's sqlite - so if you export the tracks and move to another device, the `Track.Id` are generated again (and almost guaranteed to be different).
Even if you export a track, delete it from OpenTracks, and import it again.

If you want to keep a reference (across multiple devices/ re-installations etc.) you should use `opentracks:trackid` aka `Track.getUuid()`.
This value is generated when the track is recorded and should never change (as it is exported and imported).

vsevjednom-cz, 2021-07-04:
Ok, I'll write you more about my app 🙃.

I created these AbstractActivities and another classes:
`LoginActivity` - for login (HTTP POST). It returns a token.
`UploadActivity`, `UploadAsync`, `UploadInputStream` - for upload GPX files. This is the place where I find `Uuid` from GPX and I can save it to the MySQL at my server.
`WebActvity` - for view activities of user and of people which user follows (HTTP GET with token).

User can edit name/desc/type of activity at my website. So I want to synchronize name/desc/type of last three activities (for example) when user opens `WebActivity` for example. I'll use some HTTP POST requests for synchronize of course.
`Uuid` is unique and better - I understand. But I can't use `Uuid` for edit name/desc/type of activity. So I can use `Track.id` because I know which user is logged - from LoginActivity. There can be a problem when user use more mobile devices 🤪. It can be solved (a little) by limit for last three activities.

This is the reason why:
- I need edit name/desc/type by `Uuid` (not works).
- Or I need to know `Track.id` from GPX file.

dennisguse, 2021-07-04:
Understood.
You want to use TrackEditActivity and this only accepts Track.Id, right?

If so, just extend TrackEditActivity to be able to use UUID as well - that is something we can integrate.
Just check if a Track.Id was passed or a TrackUUID using different parameter names.

vsevjednom-cz, 2021-07-04:
> If so, just extend TrackEditActivity to be able to use UUID as well.
Just check if a Track.Id was passed or a TrackUUID using different parameter names.

Yes, yes, yes! 🥳
Extend of TrackEditActivity to be able to use UUID - sounds amazing 🤩.

To make works something like this:
```
                UUID trackUuid = getIntent().getParcelableExtra(response);
                track = contentProviderUtils.getTrack(trackUuid);
                TrackUtils.updateTrack(SomeActivity.this, track, "name",
                        "type", "desc",
                        contentProviderUtils);
```

dennisguse, 2021-07-04:
Or you just do something like this:
```
Track  track = contentProviderUtils.getTrack(trackUuid);
            Intent intent = IntentUtils.newIntent(this, TrackEditActivity.class)
                    .putExtra(TrackEditActivity.EXTRA_TRACK_ID, track.getId());
            startActivity(intent);
```

vsevjednom-cz, 2021-07-04:
`Cannot resolve method 'getTrack(java.lang.String)'` 🤪
But I need update name/desc/type automatically 🙃.

dennisguse, 2021-07-05:
@vsevjednom-cz Please share some of your code.
I am good at guessing, but not that good :zany_face:

dennisguse, 2021-07-05:
```
    public Track getTrack(@NonNull UUID trackUUID) {
```

You will have to convert the String to an actual UUID.

vsevjednom-cz, 2021-07-05:
🙃 🙃 🙃
> Please share some of your code.

I want to update name/desc/type from my website for the last synchronized activity: `getNameFromWeb(token);`
`token` is user's token from my `LoginActivity`.

Method `getNameFromWeb` looks like this:
```
    private void getNameFromWeb(String token) {

        contentProviderUtils = new ContentProviderUtils(this);
        String url="https://myweb/updatename.php";
        RequestQueue requestQueue = Volley.newRequestQueue(this);

        StringRequest sr = new StringRequest(Request.Method.POST, url, new Response.Listener<String>() {
            @Override
            public void onResponse(String response) {

// response is the last synchronized Uuid at my web

                UUID trackId = getIntent().getParcelableExtra(response); // ???
                contentProviderUtils = new ContentProviderUtils(getApplicationContext());
                track = contentProviderUtils.getTrack(trackId); // crash 🤪
                TrackUtils.updateTrack(WebActivity.this, track, "name",
                        "type", "desc",
                        contentProviderUtils);

...
            @Override
            protected Map<String, String> getParams() {
                Map<String, String> params = new HashMap<String, String>();
                params.put("token", token);

                return params;
            }
...
        requestQueue.getCache().clear();
        requestQueue.add(sr);
...
```
It probably works with Track.id but I don't know Track.id from GPX file 🤪.

dennisguse, 2021-07-05:
Change:
```
                UUID trackId = UUID.fromString("whatever your test uuid is ");
``

or even better:
```
                UUID trackId = UUID.fromString(response);  // but I guess your response contains more than the UUID, right?
```

PS: that is not really an OpenTracks problem, but more an Java/ Android knowledge issue...
I would recommend at least briefly reading some books/ blogs about this.
Don't get me wrong: happy to help, but time is limited and it prevents me from doing other stuff.

vsevjednom-cz, 2021-07-05:
Wow, it works!!! 🙃 🙃 🙃
Yes, I've got more things in `response`. This is only example.

You're right. Java is new for me. This was only my misunderstanding. I'm reading and reading about Java.

So... Solved. Extend of TrackEditActivity is not necessary.

closed 2021-07-05 by vsevjednom-cz

dennisguse, 2021-07-05:
Awesome!

Here you can find some trainings: https://developer.android.com/courses
(don't know how good they are ><)

PS: Java in itself is okay, but the combination with Android's lifecycle can be very painful.

PPS: I prefer Java and there is no plan to migrate to Kotlin.
I just don't see the benefit of this language (although we use it at sometimes work :shrug:).