OpenTracksApp/OpenTracks #862: Clink announcement 🔉 for phones without TTS

vsevjednom-cz, 2021-07-01:
**Is your feature request related to a problem? Please describe.**
Phones without TTS can't use voice announcement. #763

**Describe the solution you'd like**
Some sound like sms notification every kilometer for example?

**Additional context**
Something like this: https://www.youtube.com/watch?v=TCMg0J8mfuE

closed 2021-08-30 by dennisguse

vsevjednom-cz, 2021-09-04:
I tested it while ride my mountain bike today at `Android 9.1.0`.

Voice announcement plays every 60 seconds when I set 60 minutes.

And what type of sound I must to amplify to hear voice announcement?

dennisguse, 2021-09-05:
@vsevjednom-cz by time works for me as expected. Tested with 1 and 2min. Regarding channel: the beep is played as MEDIA. Don't know if this is the best choice.

rgmf, 2021-09-05:
@vsevjednom-cz @dennisguse I think time interval settings are in seconds not in minutes, altough you can see minutes like a unit:

[image]

dennisguse, 2021-09-05:
That's my fault...

rgmf, 2021-09-06:
Fixed and push into main. You can see the commit here:

https://github.com/OpenTracksApp/OpenTracks/commit/40aab9444a9410860e3ef647b013bf8b5cf375d4

vsevjednom-cz, 2021-09-08:
> Regarding channel: the beep is played as MEDIA.

It's too quiet. I have set the maximum volume. I've got my mobile on my back when I ride my bike. Is possible to set more louder announcement? I use it as a check of turned on application.

And another idea: is possible to set announcement of autopause starts and autopause ends? I used this few years ago in Sports Tracker app.

dennisguse, 2021-09-08:
@vsevjednom-cz No clue. I guess, you need to experiment here yourself. It was literally the first time I used Android's media subsystem. Also the audio file might not be loud enough itself or it might be device dependent...

vsevjednom-cz, 2021-09-09:
I also think that the low volume of sound is because the media file is not loud. How can I select another sound file?

dennisguse, 2021-09-10:
> I also think that the low volume of sound is because the media file is not loud. How can I select another media file?

It is the only mp3 that is in OpenTracks.

vsevjednom-cz, 2021-09-10:
Wow, amazing! I already see the file `tts_fallback.mp3`. It's normalized sound but too quiet. So I "normalized" it with Waves L3 Multimaximizer. I worked in radio as presenter/DJ few years ago - so I still use some plugins for nice normalize.

You can download "normalized" version [here](https://jedu.vsevjednom.cz/tts_fallback_15db.mp3). Try to replace this file.

dennisguse, 2021-09-12:
@vsevjednom-cz just open a PR 😀 
Have you also tested if it sounds good if you listen to music on the phone using the new announcements?

vsevjednom-cz, 2021-09-12:
Yes, of course. But better is "normalize" with `Waves L2`, maybe. It's [here](jedu.vsevjednom.cz/tts_fallback_l2.mp3). Don't worry, it normalizes up to 0 db all. Mobile devices plays it without problem.
I can't open PR: `There isn’t anything to compare.`.

vsevjednom-cz, 2021-09-12:
Yes, of course. But better is "normalize" with `Waves L2`, maybe. It's [here](jedu.vsevjednom.cz/tts_fallback_l2.mp3). Don't worry, it normalizes up to 0 db all. Mobile devices play it without problem.
I can't open PR: `There isn’t anything to compare.`.

vsevjednom-cz, 2021-09-12:
Yes, of course. But better is "normalize" with `Waves L2`, maybe. It's [here](jedu.vsevjednom.cz/tts_fallback_l2.mp3). Don't worry, it normalizes up to 0 db all. Mobile devices play it without problem.
I can't open PR: `There isn’t anything to compare.`

vsevjednom-cz, 2021-09-12:
I tried out today at my mountain bike. It's much better but it wants a little bit more volume. [Here is "normalized normalized" sound](https://jedu.vsevjednom.cz/tts_fallback.mp3) 🙃.

vsevjednom-cz, 2021-09-12:
I tested it out today at my mountain bike. It's much better but it wants a little bit more volume. [Here is "normalized normalized" sound](https://jedu.vsevjednom.cz/tts_fallback.mp3) 🙃.

commit a965acf2 (2021-08-28): Voice announcements: if TTS is not available play a sound file.

Fixes #862.
commit e1c83207 (2021-08-28): Voice announcements: if TTS is not available play a sound file.

Fixes #862.
commit e33fad85 (2021-08-29): Voice announcements: if TTS is not available play a sound file.

Fixes #862.
commit fa02ba9c (2021-08-30): Voice announcements: if TTS is not available play a sound file.

Fixes #862.