OpenTracksApp/OpenTracks #880: ignore implausible GPS data

Zugschlus, 2021-07-11:
I am always frustrated when the GPS part of my phone delivers implausible data (for example one sample of latitude 0 and longitude 0 inside a run in german) and Opentracks writes out a trace file containing this implausible sample, causing the maps app to log a run that goes over thousands of miles to the equator and back. I cannot walk so large distances, and I cannot walk at the place lat 0 long 0 as it's in the middle of the ocean.

**Describe the solution you'd like**
I would like Opentracks to ignore implausible GPS data. For example, the moving speed can be a telltale sign for that. A walker or runner is unlikely to move several thousand miles in a second.

**Describe alternatives you've considered**
I'd rather not going down the road of doing manual plausibilization of my tracks.

Greetings
Marc

dennisguse, 2021-07-11:
@Zugschlus Are you sure these are GPS locations?
We also have TrackPoints that have _no location_ - more details: https://github.com/OpenTracksApp/OpenTracks/issues/639

How are you exporting?
KML/KMZ contain nowadays empty  locations aka `<gx:coords />` if those were created by the user (i.e., start recording, pausing, stop).
GPX does not contain these as a coordinate is required.

Zugschlus, 2021-07-12:
> @Zugschlus Are you sure these are GPS locations?

Nö, that was just a (probably incorrect) conclusion coming from the fact that the GPS on my Nokia 7.2 Phone sometimes reports absurd "fixes" even in other Apps.

> We also have TrackPoints that have _no location_ - more details: #639

Ah, interesting. i use Osmand+, currently Version 3,9.10.

> How are you exporting?

I call up the track and click on the map icon (top right side, left of the share icon) and then select Osmand+ in the "Open as KMZ" dialog. My 25 minute run of this morning shows up as a 12065 km trip with an average speed of 15299 km/h and a top speed of 2412812 km/h which is like eight times the speed of light. Yay, I have discovered the warp shoes!

> KML/KMZ contain nowadays empty locations aka `<gx:coords />` if those were created by the user (i.e., start recording, pausing, stop).

Start recording, run, stop. Probably some butt pauses when the phone is in the pocket. So that might be the case. and in fact an Osmand+ issue.

> GPX does not contain these as a coordinate is required.

How would I export as GPX, and can Osmand import that?

Greetings
Marc

dennisguse, 2021-07-12:
We do not include these points in GPX -> thus, the data cannot be exactly restored in OpenTracks.
So, if you have 0,0 in GPX, then something is broken with your GPS.

And OSMAnd was adjusted in March: https://github.com/osmandapp/OsmAnd/commit/80ea50ff963c7a0fc74a64612c8c4b2cf97c8b0d#diff-b0eaf9bf1f548abf17eba642d9ed3e88ce157c6506881587a919918366a95ede
So, it should understand `<gx:coords />`
But I never tested this and I could not figure how OSMAnd handles there releases...

Zugschlus, 2021-07-12:
> We do not include these points in GPX -> thus, the data cannot be exactly restored in OpenTracks.
> So, if you have 0,0 in GPX, then something is broken with your GPS.
> 
> And OSMAnd was adjusted in March: [osmandapp/OsmAnd@80ea50f#diff-b0eaf9bf1f548abf17eba642d9ed3e88ce157c6506881587a919918366a95ede](https://github.com/osmandapp/OsmAnd/commit/80ea50ff963c7a0fc74a64612c8c4b2cf97c8b0d#diff-b0eaf9bf1f548abf17eba642d9ed3e88ce157c6506881587a919918366a95ede)
> So, it should understand `<gx:coords />`
> But I never tested this and I could not figure how OSMAnd handles there releases...

Can you recommend a different mapping app that does correctly handle OpenTracks' output so that I can see whether the export is indeed ok?

As to the "something broken with your GPS", that's the reason why I am suggesting a plausibility check: Nokia is unlikely to even listen to a bug report, so the Apps need to work around. Filtering out implausible data is also a security feature.

Greetings
Marc

dennisguse, 2021-07-12:
Just try GPX - if you still see 0,0 then the GPS data is wrong.
If not, then OSMAnd does not handle `<gx:coords />` properly.

And you may use OSMDashboard.
The integration is also nicer with OpenTracks.

dennisguse, 2021-07-15:
@Zugschlus did this work for you?

Zugschlus, 2021-07-16:
I select a track, click on the Map Icon, I get one "Complete action using OpenTracks" prompt and an alternative "OpenTracks Open as KMX". When i select "Open as KMX", I get an incorrectly mapped track. When I do this again, the alternative is "Open as GPX", but the result is equally incorrect. It is not nice that the actually used format does seem to be hidden away in the dialog.

I didn't know about OSM Dashboard. Thanks for the pointer. That one works fine. So, OSMand+ is to blame.

Greetings
Marc

dennisguse, 2021-07-17:
@Zugschlus If you also see this issues with GPX, then might still be a problem in OpenTracks.
Could you send me the track, so I can take a look? dennis.guse@alumni.tu-berlin.de 

Just to be sure that is not us...

dennisguse, 2021-07-29:
@Zugschlus is this still an issue?

Zugschlus, 2021-07-30:
I havent seen that behavior in a while, but I also stopped direct track opening with Osmand+. Going through the Opentracks Dashboard works fine, including just stopping the track when the GPS is bad. I think this issue can be closed. Thanks for helping.

dennisguse, 2021-07-31:
@Zugschlus :+1: 
Please re-open if you see this issue again!

closed 2021-07-31 by dennisguse