OpenTracksApp/OpenTracks #886: Removal of AdaptiveLocationListenerPolicy?

dennisguse, 2021-07-17:
OpenTracks has the functionality to reduce the sampling rate of GPS locations if the device is stationary.
This can be activated in the `Settings -> Recording time interval` and provides the two flavors `Adapt for battery live` and `Adapt for accuracy`.
The alternative is to use a fixed time interval (actual default is `0s` aka as fast as Android provides location).

Relevant class: `AdaptiveLocationListenerPolicy`

## Assumption
The adaptive sampling rate is actually not used by our users.
If this is true, we could remove this and avoid some complexity.

:+1: if you use this functionality
:-1: if you **don't** this functionality

dfessor, 2021-07-30:
I personally don't use this functionality, but don't see a reason to remove it, if it's already built and working.
Thank you for maintaining this great piece of software. I use it with almost all of my rides!

dennisguse, 2021-07-30:
@dfessor 
Reducing complexity is pretty important.
Keeping a feature is expansive (in terms of maintenance) and it often prevents (makes it more complicated) implementing new features.
And it also it makes it makes it more difficult for new engineers to get into the code base.

Regarding this feature: I am not even sure that it is still working...
But nobody complaint, right :)

PS: nice that you like OpenTracks :sunglasses:

dfessor, 2021-07-30:
I see, thank you for explaining!
Putting this into account it's probably better to remove it and see if someone complains :-D.

commit 77f76053 (2021-08-18): Removed AdaptiveLocationListenerPolicy.
Fixes #886.
commit f67145c3 (2021-08-20): Removed AdaptiveLocationListenerPolicy.
Fixes #886.
commit 61ee4756 (2021-08-23): Removed AdaptiveLocationListenerPolicy.
Fixes #886.
PR #913 (2021-08-24): Removed AdaptiveLocationListenerPolicy.
Contains some other tiny refactorings, i.e. cleanup.

Fixes #886.
closed 2021-08-24 by dennisguse