OpenTracksApp/OpenTracks #891: Sensor: support shoe pod sensors  (RSCS) for running (distance/ speed)

dennisguse, 2021-07-25:
Time to add another class of sensors :partying_face: 

For testing, I will use a Polar Stride.

closed 2021-07-31 by dennisguse

qntnlq, 2024-05-06:
Hello Dennis, 

By any chance, is the Coros Pod supported by the commits related to this issue ? Or at least, is there any reason why pods other than the Polar Strides could not be supported ?

Thanks in advance,

dennisguse, 2024-05-06:
@qntnlq Other sensors might works as well if they support standard protocols.
For the Coro Pod, this is not documented on there webpage.
So, please check with them beforehand.

Known working sensors and some details can be found here: https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md#0x1814-running-cadence-and-speed

commit f02ea9fc (2021-07-29): Sensor: adding Running Speed and Cadence.

Fixes #891.
commit c99bc448 (2021-07-31): Sensor: adding Running Speed and Cadence.

Fixes #891.
PR #892 (2021-07-31): Sensor: adding Running Speed and Cadence.
Fixes #891.

Not yet properly tested (just implemented by specification).
Will be `forced pushed`.
commit 242147e4 (2021-07-31): Sensor: adding Running Speed and Cadence.

Fixes #891.