OpenTracksApp/OpenTracks #905: Bindings in red color in Android Studio

vsevjednom-cz, 2021-08-05:
**Describe the bug**
Android Studio shows `*Binding` in red color. Everything still works without problem 🤪.

**To Reproduce**
1. Get project version: https://github.com/OpenTracksApp/OpenTracks
2. Clic to the AboutActivity for example.
3. `de.dennisguse.opentracks.databinding` and `AboutBinding` is in red color with  this error: `Package not found: de.dennisguse.opentracks.databinding`
4. But error is gone if you replace `JavaVersion.VERSION_11`  with `JavaVersion.VERSION_1_8` in `build.gradle` - I figured it out after too much testing.

**Technical information**
 - Device: Honor KSA-LX9
 - OS: Android
 - Version: v3.19.0

Or is this something problem on my side?

dennisguse, 2021-08-05:
I don't know/
Somehow AndroidStudio is not figuring (somtimes) that this code is autogenerated...

You can try to Build -> Rebuild Project.

I also have this sometime :man_shrugging:

closed 2021-08-05 by dennisguse

vsevjednom-cz, 2021-08-16:
This is some problem in Android studio with Java 11: https://issuetracker.google.com/issues/180946610 🤪

rgmf, 2021-08-16:
> This is some problem in Android studio with Java 11: https://issuetracker.google.com/issues/180946610 zany_face

Same issue with Android Studio Arctic Fox 2020.3.1... so waiting for a new Studio release. It's not a problem for developing but an it's a little bit uncomfortable.

vsevjednom-cz, 2021-09-04:
It's fixed in version 2020.3.1.24~focal+0 in Ubuntu 🥳.