OpenTracksApp/OpenTracks #907: App crashes and cannot resume track

vsevjednom-cz, 2021-08-06:
**Describe the bug**
App crashes when tracking after tap to "desktop" and take the photo. I can reproduce app crash again and again with phone connected to the PC. Only in activity where was first crash. So I can see the error fortunately:
`E/pentracks.debu: [qarth_debug:]  get PatchStore::createDisableExceptionQarthFile method fail.
E/AndroidRuntime: FATAL EXCEPTION: main
    java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.services.sensors.BluetoothRemoteSensorManager.reset()' on a null object reference`
I don't use Bluetooth. And I can't stop this activity (and save GPX).

**To Reproduce**
It happens sometimes. I can't reproduce it.
1. Start tracking.
2. After some time tap to "desktop button" (bottom center button) and take the photo.
3. App crashes and rerun.
4. Open your last activity.
5. Tap to the menu and `Resume track`.
6. App crashes again and again.

**Technical information**
 - Device: Honor KSA-LX9
 - OS: Android
 - Version v3.19.0

vsevjednom-cz, 2021-08-06:
`Export all` crashes too 🤪.

```
E/pentracks.debu: [qarth_debug:]  get PatchStore::createDisableExceptionQarthFile method fail.
E/AndroidRuntime: FATAL EXCEPTION: main
    java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks.debug/de.dennisguse.opentracks.io.file.exporter.ExportActivity}: java.lang.NullPointerException: Missing required view with ID: de.dennisguse.opentracks.debug:id/toolbar
```

dennisguse, 2021-08-06:
@vsevjednom-cz could you test this? 

1. Duplicate of #901?
Should be fixed by #904; can you test this?

2. Duplicate of #902?
Regarding: is this happening with v3.19.0 or most recent main - as on main it should be fixed.

PS: Just open another new issue if you find different problems - they don't cost anything. ;)

vsevjednom-cz, 2021-08-07:
Yes! [Bugfix: reset TrackRecordingManager properly. #904](https://github.com/OpenTracksApp/OpenTracks/pull/904) fixes it! 🙃

closed 2021-08-07 by vsevjednom-cz