OpenTracksApp/OpenTracks #908: Crash when attempting to export all tracks from settings

dobey, 2021-08-12:
When attempting to export my tracks from Settings, the following crash happens upon clicking the _Export All_ button. The export path is configured and exists, but remains empty due to the crash.

```
08-12 11:19:43.618  5031  5031 E AndroidRuntime: FATAL EXCEPTION: main
08-12 11:19:43.618  5031  5031 E AndroidRuntime: Process: de.dennisguse.opentracks, PID: 5031
08-12 11:19:43.618  5031  5031 E AndroidRuntime: java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.io.file.exporter.ExportActivity}: java.lang.NullPointerException: Missing required view with ID: de.dennisguse.opentracks:id/toolbar
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3270)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3409)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2016)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:107)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:214)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7356)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:940)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: Caused by: java.lang.NullPointerException: Missing required view with ID: de.dennisguse.opentracks:id/toolbar
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at de.dennisguse.opentracks.databinding.ToolbarBinding.bind(ToolbarBinding.java:59)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at de.dennisguse.opentracks.databinding.ExportActivityBinding.bind(ExportActivityBinding.java:348)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at de.dennisguse.opentracks.databinding.ExportActivityBinding.inflate(ExportActivityBinding.java:178)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at de.dennisguse.opentracks.databinding.ExportActivityBinding.inflate(ExportActivityBinding.java:168)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at de.dennisguse.opentracks.io.file.exporter.ExportActivity.onCreate(ExportActivity.java:136)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.Activity.performCreate(Activity.java:7824)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.Activity.performCreate(Activity.java:7813)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1306)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3245)
08-12 11:19:43.618  5031  5031 E AndroidRuntime: 	... 11 more
08-12 11:19:43.620  1223  3908 W ActivityTaskManager:   Force finishing activity de.dennisguse.opentracks/.io.file.exporter.ExportActivity
08-12 11:19:43.620  1223  5069 I DropBoxManagerService: add tag=data_app_crash isTagEnabled=true flags=0x2
08-12 11:19:43.625  1223  1271 I ActivityManager: Showing crash dialog for package de.dennisguse.opentracks u0

```

**Technical information**
 - Device: Pixel 2 (walleye)
 - OS: Lineage+microG 17.1
 - Version v3.19.0 via F-Droid

rgmf, 2021-08-12:
It's already fixed but we have to wait for the next release. Sorry.
Thank you for your report and opening this issue ;)
PS/ Here you have more information (it's an already closed issue): https://github.com/OpenTracksApp/OpenTracks/issues/902

dobey, 2021-08-12:
Ah, I missed seeing that somehow when I was looking to see if it had been fixed yet. Thanks.

closed 2021-08-12 by dobey