OpenTracksApp/OpenTracks #909: Crash on GPX export

excodger, 2021-08-14:
**Bug description**
With the latest version of OpenTracks from F-droid, everytime I attempt to export tracks the app crashes. No settings were changed before this bug occurred.

**To Reproduce**
1. Start OpenTracks
2. Go to Settings and scroll down to Tracks section
3. Tap "Export all"
4. App crashes and restarts

I've restarted the device and tried the steps above. Also cleared cache and repeated the same steps. Also tried using a different output folder.

**Technical information**
 - Device: Motorola Moto X4
 - OS: LineageOS 17.1
 - Version: v3.19.0

rgmf, 2021-08-14:
This bug is already fixed but we have to wait for the next release.

Anyway, thank you for opening this issue ;)

dennisguse, 2021-08-16:
Fixed in v3.19.1

closed 2021-08-16 by dennisguse