OpenTracksApp/OpenTracks #911: "Continue" recording only works after second time

mega-stoffel, 2021-08-15:
**Describe the bug**
Restarting / continuing a stopped track doesn't continue the recording. But it falls back to the list of all existing tracks.
Opening the track a second time and continue it again - then it works.

**To Reproduce**
1. Start recording, record some time and then stop your recording.
2. On the list of all available tracks select this track
3. On the detail view of this track, click on the three dots in the top right corner.
4. Click "Track fortsetzen" (continue recording).
5. Now the app behaves a bit strange, shows some Splashscreen for a short moment and the it goes back to the list of all recordings.
6. Open the track again and hit "continue" a second time - now it works and the recording continues.

If applicable:
-

**Technical information**
 - Device: Fairphone 3
 - OS: Android 10
 - Version 3.19 from f-droid store

rgmf, 2021-08-15:
@mega-stoffel it's already fixed. De have to wait for the next release.

Anyway, thank you so much to inform about it ;)

dennisguse, 2021-08-15:
@rgmf Anything that prevents us from releasing v3.19.1 (aka fixing this and #901)?
i pushed a fix for a race condition (occurred on CI) due to unregistering delayed unregistering from LocationManager.

rgmf, 2021-08-15:
> @rgmf Anything that prevents us from releasing v3.19.1 (aka fixing this and #901)?
> i pushed a fix for a race condition (occurred on CI) due to unregistering delayed unregistering from LocationManager.

@dennisguse for me it's ready for v3.19.1 (I've been using main branch for a week and all looks work well).

dennisguse, 2021-08-15:
@rgmf Do you want to release it? :partying_face: 
Everything should be in `RELEASE.sh`.

rgmf, 2021-08-15:
> @rgmf Do you want to release it? partying_face
> Everything should be in `RELEASE.sh`.

Okay, here I go :blush:

rgmf, 2021-08-15:
@dennisguse I think it's done ;)

dennisguse, 2021-08-16:
Fixed in v3.19.1

closed 2021-08-16 by dennisguse