OpenTracksApp/OpenTracks #916: Edit track while recording is not working.

rgmf, 2021-08-21:
I think this issue was fixed some time ago so we should fix it again and add tests cover.

**To Reproduce**
1. Start a new recording.
2. While recording go to menu (three dots) and select edit.
3. Change name and activity type, for example.
4. Save the changes and back to recording layout.
5. Stop recording.
6. Name and activity is not the edited one.

**Technical information**
 - Device: any
 - OS: any
 - Version: v3.19.0 and v3.19.1

dennisguse, 2021-08-21:
Interestingly, while recording, the UI is showing the updated info.
And those should be loaded from the db, so we are overwriting it while finishing the recording?

rgmf, 2021-08-21:
> Interestingly, while recording, the UI is showing the updated info.
> And those should be loaded from the db, so we are overwriting it while finishing the recording?

Yes, it seems like this.

dennisguse, 2021-08-21:
The bug is in TrackRecordingManager as this one now caches the Track instead of loading it every time from the database.
The track is only loaded when it is resumed or when it is pushed to the UI....
I did this "performance improvement"...

rgmf, 2021-08-21:
> The bug is in TrackRecordingManager as this one now caches the Track instead of loading it every time from the database.
> The track is only loaded when it is resumed or when it is pushed to the UI....
> I did this "performance improvement"...

I got this for tomorrow but it you can and want go ahead and fix it ;)

commit d6badd8c (2021-08-22): TrackRecordingManager only updates track's statistics every time a new track point is inserted. Fixes #916.
commit 13380afe (2021-08-22): TrackRecordingManager only updates track's statistics every time a new track point is inserted. Fixes #916.
closed 2021-08-22 by rgmf