OpenTracksApp/OpenTracks #924: CooSpo Bluetooth sensor no longer detected.

badger40, 2021-08-25:
CooSpo cadence and speed sensor (model: BK467) is no longer detected when scanning for Bluetooth sensor in the app.
Go to:
settings -
Bluetooth -
Cycling distance and speed -
Scan for sensor and it's not found.

Used nRF toolbox app to test in sensor is faulty or dead battery.  Sensor is picked up and working fine. I've not used opentracks for a while so not sure when this issue started.

Open Tracks version name: V3.19.1
Version code:4396

Phone: Poco F1
Miui version: global 12.0.3
Android version: 10

Kind regards.

dennisguse, 2021-08-25:
I just tested using LineageOS 17.1 with main (v3.20.+; not yet released) and it works.
Recent changes:
* added new sensor: https://github.com/OpenTracksApp/OpenTracks/commit/242147e49122563d35eaf0105fbc6bf79f75edc3
* upgrade for Android12 (changing Bluetooth permissions):
  * https://github.com/OpenTracksApp/OpenTracks/commit/26414b0c7ccfe322db7e49183260e31aa4440635
  * https://github.com/OpenTracksApp/OpenTracks/commit/71c385a1a9e5de29fff498ca62aca6f82e05701f 
  * https://github.com/OpenTracksApp/OpenTracks/commit/afbd3aa9973abaf7b4c18baf7ebf0c9743fe05ee

dennisguse, 2021-08-25:
@badger40 can you test with v3.18.5 as this was before these changes?