OpenTracksApp/OpenTracks #930: Speed sensor: speed is never 0km/h

rgmf, 2021-08-30:
**Describe the bug**
When you are recording a track with a cycling sensor speed and you stop at a semaphore, for example, the speed data never is 0km/h so moving time go on.

I guess the speed I see in this field is the last the sensor send to OT.

**To Reproduce**
1. Start a new recording with a speed sensor.
2. Go ahead and stop.
3. You'll see the speed going down until some speed like 4km/h, for instance.
4. When you are completely stopped the speed is not 0km/h but the last sensor sended to OT.

**Technical information**
 - Device: Pixel 2 XL
 - OS: LineageOS 17.1
 - Version v3.19.x and v3.20.0 (I couldn't test it in older versions)

dennisguse, 2021-08-30:
Nice finding - I never noticed that.
I thought we use SensorData.isRecent() to determine if we got "fresh" data (Default timeout is 5s; don't know why I came up with 5s. It was for HR sensors).
Afterwards we do some computation in TrackStatisticsUpdater to smoothen the speed (only done when new data arrives).
So, this might be the reason (best guess, did not check).

rgmf, 2021-08-30:
> Nice finding - I never noticed that.
> I thought we use SensorData.isRecent() to determine if we got "fresh" data (Default timeout is 5s; don't know why I came up with 5s. It was for HR sensors). [...]

I see.

This will be a great issue for me so I can look into this part of OT that is unknown for me 😄

dennisguse, 2021-08-30:
It is not the TrackStatisticsUpdater, but I guess StatisticDataBuilder.
It checks the SensorData.isRecent() and if not uses the GPS value (and that is already old likely containing movement).

I guess, SensorData.isRecent() should be removed and should be handled internally by SensorData.getValue() aka return default value if the data is too old.

rgmf, 2021-09-18:
2021-09-18 Update:

Same issue with Speed GPS... When you are using GPS for speed and distance (not sensors connected) and you stop pedaling GPS speed is the last GPS valid speed but it should be 0.

commit d9bf7f75 (2021-09-18): Cadence set to 0 when connected but user is not pedaling. Fixes #931.
Speed set to 0 when connected but last value is outdated. Fixes #930.
Bug: overflow when reset sensors (cadence and speed). Fixes #833.
Moving time stops when is not moving (the last track point is not moving) and restarts when last track point is moving.
commit d99fb5ea (2021-09-21): Cadence set to 0 when connected but user is not pedaling. Fixes #931.
Speed set to 0 when connected but last value is outdated. Fixes #930.
Bug: overflow when reset sensors (cadence and speed). Fixes #833.
Moving time stops when is not moving (the last track point is not moving) and restarts when last track point is moving.
closed 2021-09-21 by rgmf