OpenTracksApp/OpenTracks #931: Cadence 0rpm instead of None

rgmf, 2021-08-30:
**Is your feature request related to a problem? Please describe.**
I'd prefer to see 0rpm instead of None when I stop pedaling my bike.

**Describe the solution you'd like**
When cadence sensor is connected never shows "None" like a value but "0 rpm".

dennisguse, 2021-08-30:
Should it be value_unknown?

rgmf, 2021-08-30:
> Should it be value_unknown?

IMHO I should be value_unknown when cadence sensor is not connected and 0rpm if it's connected.

But I don't know very well how sensors work. Does cadence sensor send 0 when cyclist is not pedaling? If so, I don't understand where None value comes from.

dennisguse, 2021-08-30:
You are right that are two cases.

Regarding the data: those sensors only send data when they measured something (like the wheel rotated).
If nothing is happening, then they don't send data.
So, the timeout occurs OpenTrack invalidates the data (5s timeout) and then the data is null resulting and `StringUtils.formatCadence()` returns None.

So, the invalidation should actually set them 0 and on disconnect to null.

commit d9bf7f75 (2021-09-18): Cadence set to 0 when connected but user is not pedaling. Fixes #931.
Speed set to 0 when connected but last value is outdated. Fixes #930.
Bug: overflow when reset sensors (cadence and speed). Fixes #833.
Moving time stops when is not moving (the last track point is not moving) and restarts when last track point is moving.
commit d99fb5ea (2021-09-21): Cadence set to 0 when connected but user is not pedaling. Fixes #931.
Speed set to 0 when connected but last value is outdated. Fixes #930.
Bug: overflow when reset sensors (cadence and speed). Fixes #833.
Moving time stops when is not moving (the last track point is not moving) and restarts when last track point is moving.
closed 2021-09-21 by rgmf