OpenTracksApp/OpenTracks #936: Multi-column stats

JochenLutz, 2021-09-03:
**Is your feature request related to a problem? Please describe.**
When cycling, the most important stat for me is current speed. While the new customizable layout really helps to quickly capture the current speed, having the speed value span all columns (with increase font size), would further improve cycling safety.

**Describe the solution you'd like**
In the “Customize your recording layout“, in the settings of a stat item, add an option to make that item span all columns. Items spanning all columns should get bigger font — font size increase between new multi-column and primary items should be comparable to increase between primary and secondary items.

**Additional context**
[image]

rgmf, 2021-09-03:
It looks like a feasible improvement and I think it'd be nice.

Only for clarification: you expect three sizes for data fields, don't you?:
- Small for secondary.
- Big for primary.
- Extra-big for spanned.

JochenLutz, 2021-09-03:
Exactly.

closed 2024-12-29 by dennisguse