OpenTracksApp/OpenTracks #937: Feature Request: Treadmill Option

convolutionbbs, 2021-09-05:
I was wondering if anybody else would be interested in using opentracks to log treadmill runs.  I would like to use it to log my HR via BT and have the ability to edit the distance after the recording.  This would allow the stats to be included in total metrics.

dennisguse, 2021-09-05:
To log sensor data without GPS is handled by #500. Should be ready in October. 
Adding data by hand is not yet on the roadmap.

Does the treadmill provide data via Bluetooth?

convolutionbbs, 2021-09-05:
Thanks for the reply.  No, my treadmill doesn't have bluetooth.  I have a HR monitor that is bluetooth and it works really well with OpenTracks.  I was thinking that the user would connect the HR monitor and begin and log a workout just like if I ran outside.  After I was done, I could hit the edit button and manually type in how long the treadmill thought I ran.  That way I could see the HR data and the long term stats would capture the distance.

dennisguse, 2021-09-06:
That's doable. 
What must be done is to set the total distance as the distance of the final TrackPoint and then re-create the TrackStatistics. 

And it could be done for recordings for which 0 distance was recorded.

Intermediate solution: use of a shoe-based running sensor.
Then the overall distance will taken from that there.
Until #500 is resolved only the one TrackPoint is stored (ie. no intermediate values from HR/ shoe sensor).

brianjmurrell, 2022-03-08:
This would be really cool.  A lot of my workouts are on the treadmill.

MrS1R1US, 2023-01-26:
> What must be done is to set the total distance as the distance of the final TrackPoint and then re-create the TrackStatistics.

I don't understand how to do it. Where to click? Tell me please

dennisguse, 2023-01-29:
@MrS1R1US this functionality is not implemented.
It is just the description how it could be done.

closed 2024-12-29 by dennisguse