OpenTracksApp/OpenTracks #951: gps on/off/fix icon

midros, 2021-09-14:
I think it is better to have a gps icon to turn the gps on / off as well as a clear gps fix indication
It used to be like this in the previous versions but now is removed. 
The way it is right now without gps fix indication you may start recording but there is no purpose if the gps position is not ready

dennisguse, 2021-09-14:
Can you share a screenshot about what you mean?

midros, 2021-09-14:
[image]
[image]
these are screenshots from a previous version
the arrow shows the icon for turning the gps on-off
and the icon changes when the gps is fixed

dennisguse, 2021-09-14:
Mhhhh.... I just tested with v3.21.1 and the icon works as expected (as in previous versions). Do you have some more details?

midros, 2021-09-14:
ok now I see why the confusion

I have the f-droid version 3.20.1 and there is no gps icon, so I thought it was removed for some reason, but obviously this is not the case. (a bug maybe?)

here is a real screenshot of my app

[image]

dennisguse, 2021-09-14:
I see :)
If the icon is shown or not is managed by android depending on the available space (e.g., screen size and desired object size depending on the zoom level of the user). If there is not enough space the remaining icons are moved into the menu (max 3 icons are shown). 

Long story short: if the icon is hidden, try to change androids zoom level (don't know how it is actually called, it is somewhere in androids settings).

midros, 2021-09-14:
and of course you are right but how could I know?

I changed the zoom level and the icon is back

so the change happened after last android update and not opentracks update

thanks a lot

dennisguse, 2021-09-14:
Don't worry. Cool that ist works again :)

closed 2021-09-14 by dennisguse