OpenTracksApp/OpenTracks #952: Nightly improvements

dennisguse, 2021-09-17:
* install git
* use black icon background
* (also for debug) change background Color of splash screen
* append nightly to app name

pstorch, 2021-11-14:
I've got the build to run on my little vserver and setup a cronjob for 1 a.m. to check for new commits and build a new nightly apk.
Let's see if it is working.
In case someone has already installed the first test nightly from my repo, you need to uninstall it first. I had to change the signing key.

Repo (still beta): https://fdroid.storchp.de/fdroid/repo/

dennisguse, 2021-11-14:
@pstorch :heart_eyes: 
Just installed the nightly on my phone (in parallel the other two installations ><).
Now I am curiously waiting for the next just released version...

pstorch, 2021-11-16:
First successful build via cronjob last night 🎉

dennisguse, 2021-11-16:
@pstorch Got the update in the morning!
Nice work;

dennisguse, 2021-11-16:
Now we have a black background for the launcher icon, so I distinguish all three variants :D
Staying tuned for the new nightly build :partying_face:

closed 2021-11-16 by dennisguse

pstorch, 2021-11-17:
I got a black icon 🎉

dennisguse, 2021-11-19:
@pstorch PR for the webpage welcome :D

commit 6b84d3b9 (2021-11-16): Nightly build: icon has now a black background.

Fixes #952.