OpenTracksApp/OpenTracks #955: Show gradient of uphill/downhill

vsevjednom-cz, 2021-09-23:
**Is your feature request related to a problem? Please describe.**
My friend has good idea: show current gradient of uphill/downhill.

**Describe the solution you'd like**
Calculation while recording. It's similar like speed count. I'm using for calculation something like this in PHP: round(100*($altitude-$lastaltitude)/$distance)
Result is 10 % for example - it's hard uphill for bike 🚲 but I love it!

**Additional context**
Another column in customize recording layout.

dennisguse, 2021-09-23:
Nice idea!

One disclaimer: absolute altitude via GPS is not that precise (contains quite some noise).
It might be better to use the barometric sensor (or some combination).

The good old MyTracks contained some max/min grade, but I don't remember how this was derived - might be by PlayServices API.
Could be worth checking the old code.

dennisguse, 2021-09-23:
One more thing: barometric data is already relative to the previous stored location (as the counter is set to 0m on storing it). 

But it may contain values for DESCENT and ASCENT at the same time as both counters are stored to track changes in both directions between stored locations/TrackPoints.
And we do not store the order in which ascent/descent occurred (so far).
It's more of an edge case :)

rgmf, 2021-09-24:
I love this feature and it'd great to see it in OpenTracks.

It also looks a good challenge :smile:

vsevjednom-cz, 2021-09-24:
Yep, altitude data via GPS are not precise I know. You can use barometric sensor of course. But my mobile phone doesn't have barometric sensor as you know from my issue #756 (gain and loss still now works 🙄). Some combination sounds good 😋!

vsevjednom-cz, 2021-09-24:
Yep, altitude data via GPS are not precise I know. You can use barometric sensor of course. But my mobile phone doesn't have barometric sensor as you know from my issue #756 (gain and loss still not works 🙄). Some combination sounds good 😋!

vsevjednom-cz, 2021-09-27:
I tried gradient/slope calculation in PHP. GPS elevation values are sometimes weird 🤪. Better will be check gradient/slope for current elevation versus previous 3 (or 5) values for example. It could eliminate bounces.

Like this:
`round(100*($altitude-$lastaltitude3)/($distance+$lastdistance1+$lastdistance2+$lastdistance3)`

(I wrote this code from my mind, it's makes sense I hope.)

dennisguse, 2021-09-28:
@vsevjednom-cz You may take a look at https://www.gpsvisualizer.com/tutorials/elevation_gain.html
They explain nicely how to filter/ smooth GPS elevation values to compute gain/ loss.
And if we have gain/ loss, the gradient will available almost out of the box.

closed 2024-11-21 by dennisguse