OpenTracksApp/OpenTracks #957: Optimisations for OLED

dennisguse, 2021-09-24:
**Is your feature request related to a problem? Please describe.**
OLED screens do not consume energy when presenting black content (thus they can be used for always on displays).

Thus, we could implement a theme for OpenTracks that uses black as background colour. We may be able to optimise out further by and hiding non-necessary elements (e.g., icons).

**Describe the solution you'd like**
Implement a configurable option to replace the dark theme with a black theme.

PS: Strictly necessary is this only while recording, but a theme might be easier than one custom activity.

**Describe alternatives you've considered**
Just use the dark theme.

Original idea: https://github.com/OpenTracksApp/OpenTracks/discussions/945

dennisguse, 2021-11-29:
An example showing use of setTheme in every activity is referenced here: https://github.com/OpenTracksApp/OpenTracks/issues/1038#issuecomment-981166777

dennisguse, 2022-01-14:
@ae5960e8-a6fc-491f-b252-898ecf59af95 if you have some time, go for it.
The F-Droid client has this kind of feature and shouldn't be too difficult to adapt.

closed 2022-03-03 by rgmf

ae5960e8-a6fc-491f-b252-898ecf59af95, 2022-03-15:
Can't wait for the release!

dennisguse, 2022-03-15:
@ae5960e8-a6fc-491f-b252-898ecf59af95 it is in v3.26.0 (released yesterday), but I forgot to put it into the release notes.

@rgmf sorry

rgmf, 2022-03-15:
> @ae5960e8-a6fc-491f-b252-898ecf59af95 it is in v3.26.0 (released yesterday), but I forgot to put it into the release notes.
> 
> @rgmf sorry

No problem, I didn't remember of that either :smile:

commit d6c25bf0 (2022-03-03): Migration to Material Components theme and style.
New night theme mainly white on black.
Fixes #1122.
Fixes #957.
commit 79e45f7a (2024-04-18): Migration to Material Components theme and style.
New night theme mainly white on black.
Fixes #1122.
Fixes #957.