OpenTracksApp/OpenTracks #961: Refactor use of SharedPreferences

dennisguse, 2021-09-26:
SharedPreferences can be referenced as a static field.
This would simplify the handling them, as we don't need to pass the sharedpreferences that we got from each individual context.

The solution is to just keep the sharedpreferences as a static field in PreferencesUtils and init it if it is null.
Sharedpreferences can then be removed everywhere.

For OSMDashboard it was done a while ago: https://github.com/OpenTracksApp/OSMDashboard/pull/102

dennisguse, 2021-09-28:
@rgmf I am working on this and I should be done until Thursday.
But the changeset is rather large - I hope you don't have too many changes on your machine...

rgmf, 2021-09-29:
> @rgmf I am working on this and I should be done until Thursday. But the changeset is rather large - I hope you don't have too many changes on your machine...

No problem, I've not done anything for now (still a little busy). In a few days I'll be more free and I'm thinking what to do: profiles for customization layout or re-organize settings (#960)... What shall we do? What do you think is more important?

dennisguse, 2021-09-29:
@rgmf Whatever you is more interesting for you ;)

commit 3a8b8b2a (2021-09-28): Cleanup: SharedPreference as static variable.

Fixes #961.
commit c4549f78 (2021-09-29): Cleanup: SharedPreference as static variable.

Fixes #961.
commit 607e1020 (2021-10-10): Cleanup: SharedPreference as static variable.

Fixes #961.
commit eac3d4b3 (2021-10-13): Cleanup: SharedPreference as static variable.

Fixes #961.