OpenTracksApp/OpenTracks #967: Track recording stops when screen is locked (it seems)

sjvudp, 2021-10-02:
**Describe the bug**
This problem is the same as described in https://github.com/mendhak/gpslogger/issues/916:

Recording positions stops when the screen is locked.
As I carry my phone in a pocket or a backpack, the screen is locked to avoid random input.

**To Reproduce**
1. Start track recording
2. Activate lock screen
3. Move...
4. Stop recording

The track being recorded has only one or two track points.

**Technical information**
 - Device: Huawei P10 Lite
 - OS: Android
 - Version: 8.0.0

dennisguse, 2021-10-02:
It is the power saving. Huawei is doing very ugly things here.
You can check if this helps: https://dontkillmyapp.com/huawei

Bad thing: We cannot do anything about this.

PS/ Huawei is the worst in this domain :shrug:

closed 2021-10-02 by dennisguse

sjvudp, 2021-10-05:
It's not so easy: I've allowed "background activity", and I've enabled a warning if the app needs too much power (but I got none). Still when the app stops recording, it is still running (I'd say it was not terminated by the OS).

dennisguse, 2021-10-05:
Still running just means that was not killed, but the application could be in a hibernated state in which it is not woken up.
This might, for example, be applied if the battery saver is active (or the vendor modified Android so it would save more battery like Huawei).
Usually applications get only killed if not enough memory is available - but that rarely happens.

For me, it is difficult to check this (lack of Huawei devices) and such issues also prevent that I would buy Huawei devices.

So far, we had some trouble with power saving, but it always device misconfiguration/ vendor issues.
While recording we get a WakeLock and Android should respect that.
Not too much more that OpenTracks can do.

sjvudp, 2021-10-06:
I'm no developer, and I haven't looked closely at the app, but I had the impression that an app doing background things needs special programming and metadata saying that it wants a permission for background processing. I could be wrong, but maybe check.
Some sources I found:
https://developer.android.com/training/location/request-updates#request-background-location
https://developers.google.com/location-context/fused-location-provider
https://developer.android.com/reference/android/location/LocationManager
https://github.com/codepath/android_guides/issues/220

dennisguse, 2021-10-06:
We have this in place - just that Huawei does not respect it:
* we start a Foreground Service (incl. notification)
* we get a WakeLock
And we have the required configuration in the manifest.

Just some vendors believe that fine-tuning is a good idea :shrug:

IT-bob, 2022-02-11:
I think the problem is because OpenTracks asks for the location permission `only while using the app` and not `allow all the time`.
I have the same issue on OnePlus 5T (LineageOS 18.1 / Android 11), the battery optimization is disabled, but I don't have this issue with others applications, like PhoneTrack, who asks for `allow all the time` location permission.

dennisguse, 2022-02-11:
@IT-bob we are using a foreground service and thus `only while using the app` is sufficient according to the Android documentation.
Just some vendors are .....
And `allow all the time` is actually a privacy issue and also severely limited on newer Android's (just some updates every now and then).

IT-bob, 2022-02-12:
Okay, I see. Thanks

sjvudp, 2022-02-14:
> @IT-bob we are using a foreground service and thus `only while using the app` is sufficient according to the Android documentation.
Well, it may depend on usage: If the lock screen is active, is the app still a "foreground service"?
Personally I see the app more as a "background locker" rather than a "screen staring application".

IT-bob, 2022-02-14:
I just noticed something else: contrary to what I thought, it's not only related to the location. I use a speed sensor, which works well when the screen is unlocked, but I no longer receive data if I lock it.

> Just some vendors are .....
So, it's more a hardware issue than software one? I'm on LineageOS and don't use OxygenOS (OnePlus ROM).

sjvudp, 2022-03-24:
> (...) I use a speed sensor, which works well when the screen is unlocked, but I no longer receive data if I lock it.
> So, it's more a hardware issue than software one? (...)

If the app was designed to stare at the screen all of the time, while emptying the battery, it's the wrong app for me (even with a locked screen very few phones keep up for a whole day); I want to start the app, maybe use other apps in between, lock the screen and put the phone in my pocket. Only at the end of the tour I want to stop recording and see the results.
That has very little to do with hardware!

dennisguse, 2022-03-24:
@IT-bob @sjvudp it is (usually) not a hardware issue, but really just the Android OS - more precisely whatever the vendor modified to create the problem.

OpenTracks works nicely when the screen is off or locked or even on.
Even with GPS or Bluetooth sensors or without (don't know why anybody would want to do this but YOLO).

Due to the fragmentation of the Android landscape that "problem" is re-occurring every now and then, but there is nothing we can do about it.
Sorry.

Installing a proper Android OS variant might solve the issue, but that has disadvantages as well.

sjvudp, 2022-06-06:
Recently I gave OpenTracks (3.27) another try: I ran it in parallel with ape@map, comparing the results at the end.
Note that both applications were running on the same device using the same power management options.
I exported the recorded track to open it in ape@map to compare (OpenTracks cannot display the track well it seems).
The maps are basically openstreetmaps.

First OpenTracks:
[image]

Next ape@map:
[image]

Besides the huge difference in recording quality, the overall stats differ significantly also:

- ape@map: track duration is 8:13:57, total distance is 32.92km
- OpenTracks: track duration is 4:45:22, total distance is 9.14km

I think the whole open source thing does not count if the app is just unusable.

dennisguse, 2022-06-06:
@sjvudp you are free to investigate and fix the issue.
As long as I don't get my hands on a Huawei P10 Lite, I would not be able to work on this.
Moreover, why should the open-source community (incl. me) fix bugs in Huawei's Android OS?

For me, OpenTracks is a spare-time project: mainly doing it for fun.
If it works for you, that's okay for me.
If it doesn't work for you, that's also okay for me.
The source code is freely available, so have fun adjusting it to you needs (aka fixing the power management situation on your Huawei).

piotrjurga, 2022-10-05:
I get the exact same issue on two phones: Samsung A52s and Xiaomi Redmi Note 8T. On both the tracking stops when the screen is locked which makes the app pretty much unusable.