OpenTracksApp/OpenTracks #968: Filter locations using GPS speed

vlmendz, 2021-10-03:
**Is your feature request related to a problem? Please describe.**
The app does not always respect the specified distance interval to discard locations. I use the default value (10 m) and frequently the app record locations at a shorter distance, especially when I am not moving

**Describe the solution you'd like**
Optionally use GPS speed to filter locations with a speed lower than a value specified by the user (in addition to distance filter, this would allow in many cases to improve the precision of the calculation of the distance traveled and the time in movement, especially when there are long stops or many stops along the way)

**Describe additional features you've considered**
Allow optionally use GPS speed to determine maximum speed in addition to current method

dennisguse, 2021-10-03:
@vlmendz Could you share some test data (the shorter the better) to show the problem?

PS/ Speed is only estimated by GPS and might not be that useful.
It has the tendency to overestimate it - that's we filter the speed for showing in the UI.

vlmendz, 2021-10-03:
[image]
Example of long stop (aprox 30 minutes)☝️

[image]
Example of a short stop at a traffic light☝️

In my experience, using GPS speed in conjunction with distance between locations improves location filtering when you're not moving. I propose to add it as an optional feature and that the user can decide whether or not to use it depending on the behavior of their GPS device (as do the OsmAnd and Cruiser apps for example)

dennisguse, 2021-10-04:
I see and I have seen this pattern quite often while testing at home (indoors with really limited accuracy).
Could you check that the speed would be an indicator for outliers?

The other workaround for now are: 
* require stricter accuracy (default 50m)
* increase recording distance (default 10m)

vlmendz, 2021-10-04:
Yes, importing the track of the first screenshot into the GPXSee program and using a GPS speed filter with a value of 3 km/h was enough to eliminate unwanted locations and improve track statistics (the length of the track (approx 10 km in total) was reduced by around 200 m and the average speed in motion was increased by 1.5 km/h by decreasing the time in motion by a few minutes).

I tried with a distance interval of 20 m, but in my assessment there is some problem in the OpenTracks distance filtering algorithm because sometimes it records locations at distances below the set value.

vlmendz, 2021-10-04:
With my current device and for my use of OpenTracks (riding bike in the city), a suitable value would be around 3 km/h (0.833 m/s). I see from your comment that the speed filter is already programmed, it would be excellent to add in Settings>Advanced options the possibility to modify the filter value.

dennisguse, 2021-10-04:
@vlmendz Could you do this yourself?

vlmendz, 2021-10-04:
I don't have any software to develop apps for Android, besides 0 experience with Java and programming for Android.
Perhaps I could try to find the problem with the distance filter if you tell me in which file(s) it is done and which function is used to calculate the distance between points.

dennisguse, 2021-10-04:
Then don't worry; I might have some time the upcoming weekend.
Takes me about 30min :)

vlmendz, 2021-10-04:
Ok, I'll be watching for updates :)

vlmendz, 2021-10-05:
I download the source code and examining the TrackRecordingManager.onNewTrackPoint() function I think here's the cause of the erratic operation of the distance filter.
I propose you to analyze or test this modified version of the function:

    void onNewTrackPoint(TrackPoint trackPoint, Distance thresholdHorizontalAccuracy) {
       
        //TODO Figure out how to avoid loading the lastValidTrackPoint from the database
        
        TrackPoint lastValidTrackPoint = getLastValidTrackPointInCurrentSegment(trackId);

        //Storing trackPoint

        // Always insert the first segment location
        if (!currentSegmentHasTrackPoint()) {
            insertTrackPoint(trackId, trackPoint);
            lastTrackPoint = trackPoint;
            return;
        }

        Distance distanceToLastTrackLocation = trackPoint.distanceToPrevious(lastValidTrackPoint);
        if (distanceToLastTrackLocation != null) {
            if (distanceToLastTrackLocation.greaterThan(maxRecordingDistance)) {
                insertTrackPointIfNewer(trackId, lastTrackPoint);

                trackPoint.setType(TrackPoint.Type.SEGMENT_START_AUTOMATIC);
                insertTrackPoint(trackId, trackPoint);

             // The isIdle variable would no longer be required          
             // isIdle = false;

                lastTrackPoint = trackPoint;
                return;
            }
            
            /**
            * 1st modification:
            * Only add point if distance and speed filter conditions are met
            * A value of 0 in Speed.MAX_NO_MOVEMENT_SPEED disables the speed filter
            */
            if (trackPoint.hasSensorData() || (draistanceToLastTrackLocation.greaterOrEqualThan(recordingDistanceInterval) && trackPoint.isMoving())) {
                insertTrackPointIfNewer(trackId, lastTrackPoint);

                insertTrackPoint(trackId, trackPoint);

             // The isIdle variable would no longer be required
             // isIdle = false;

                lastTrackPoint = trackPoint;
                return;
            }
        }

       /**  
       * 2nd modification:
       * Remove this code segment to avoid adding points 
       * without complying with the distance or speed filter
       *
       *if (!isIdle && !trackPoint.isMoving()) {
       *    insertTrackPointIfNewer(trackId, lastTrackPoint);
       *
       *    insertTrackPoint(trackId, trackPoint);
       *
       *    isIdle = true;
       *
       *    lastTrackPoint = trackPoint;
       *    return;
       *}
       *
       *if (isIdle && trackPoint.isMoving()) {
       *    insertTrackPointIfNewer(trackId, lastTrackPoint);
       *
       *    insertTrackPoint(trackId, trackPoint);
       *
       *    isIdle = false;
       *
       *    lastTrackPoint = trackPoint;
       *    return;
       *}
       *
       * End of removed code
       */

        Log.d(TAG, "Not recording TrackPoint, idle");
        lastTrackPoint = trackPoint;
    }

 P.s. To modify the value of the speed filter I suggest putting an EditBox in "Advanced options" that allows floating point values between 0 and 15 or 20 (in km/h for metric units and mph for imperial ones)

dennisguse, 2021-10-05:
@vlmendz Yes, you found the code.
The problem is: I don't know why it was built this way, so modifying is in such cases slightly risky.
It is kind of a guessing game ;)

So, if we have good arguments why it should be removed (aka it is of no good), then we can remove it. Finding the arguments is usually the tricky part....

rgmf, 2021-10-05:
@dennisguse I've never understood very well that piece of code.

> My proposal: lets wait until 2021-10-15 if we find any reason why we have this code.
>  If there is none, we remove it.

Okay, I'm going to re-reading and look into this in the following days so I hope I can help on this.

rgmf, 2021-10-06:
@dennisguse @vlmendz I've gone to ride my bike with two smartphones: one with the code deleted and the other one with the last code in main branch and these are the results:

| (A) Phone with changes proposed by @vlmendz | (B) Phone with main branch |
|-|-|
|44,86km|44,75km|
|Moving time: 1:39:02|Moving time: 1:34:31|

I stopped riding about 5 minutes and (B) behaved better than (A). Moving time in (A) was stopped but don't in (B). Also, moving time in (B) got crazy (it went backward and forward quickly).

I don't know why, but code deleted seem important.

Lastly, I don't see in OSMDashboard the behaviour pointed out by @vlmendz in the path. Maybe because I've been in places with very good GPS signal. Seemed that OT didn't create points while I wasn't riding.

Anyway, I'll continue testing it and trying to understand what really the code does. Also, I'll trying to reproduce @vlmendz issue.

rgmf, 2021-10-07:
@vlmendz I've just found out an error in my coding. I wrote this:

```java
if (trackPoint.hasSensorData() || (distanceToLastTrackLocation.greaterOrEqualThan(recordingDistanceInterval)) && trackPoint.isMoving()) {
```

Instead of:

```java
if (trackPoint.hasSensorData() || (distanceToLastTrackLocation.greaterOrEqualThan(recordingDistanceInterval) && trackPoint.isMoving())) {
```

:sob: :sob: :sob:

So, everything I said doesn't have any sense... Sorry!

rgmf, 2021-10-07:
> @vlmendz @rgmf Could you just create a PR for this? Easier to test :D

Here it is: #972

dennisguse, 2021-10-07:
I did some cleanup on the #971 in the `TrackRecordingManager` (contained too many TODOs).
I anyhow needed to simplify it to finally fix #969 completely.
TLDR: for actual changes I would recommend using #971 and apply the fixes of #972 there.
(@rgmf sorry for the duplicate work; I was trying to find a "nice" solution for the distance sensor problem)

vlmendz, 2021-10-09:
To check filtering behavior, I recorded a track with the Cruiser app, following the same route as the one shown above and using the same device (Redmi Note 7).

Signal reception conditions:
Not the best. Urban area with many buildings and large trees.

👇Values used in Cruiser filters:
[image]

Results:
The results obtained were very good. The visual examination of the track does not show unwanted positions or "knots" at the stops (1010 points recorded with OpenTracks vs 686 with Cruiser).
The analysis of the length of the track segments yielded almost perfect results; only one segment measured less than 10m (9,943m, I suspect this is due to a small inaccuracy in the method used by Cruiser to calculate distances).

[image]
☝️ Track segment classified by length (note that a single segment below 10m)

[image]
☝️Only one segment not meet distance filter value by 57mm (probably small inaccuracy in distance calculation)

[image]
☝️Fragments of tracks. Completely in green the one recorded with Cruiser (all segments length>=10). Almost completely in red the recorded with OpenTracks. In the area within the blue circle a long stop was made on both tracks.  An interesting fact is that on the moment of the recording with Cruiser the reception conditions were worse (in several places of the track the locations left a little off the street)

Conclusions:
The use of simple filters of speed and distance with adequate values allows to significantly improve the quality of recording of tracks using low end GPS receivers and in areas without the best reception conditions.

vlmendz, 2021-10-10:
I imported in OpenTracks the track recorded with Cruiser and noticed that the calculation of the moving time fails:

[image]

Importing the same track into other apps I got the following:

Total time: 1:28:40
Moving time:
OsmAnd: 38:02
OruxMaps: 38:00
GPX Viewer: 38:02
GPXSee: 35:39
(the latter is the most correct. This program uses during import the value of the speed filter set by the user. By putting the same value used during the recording (3km/h), in the speed/time graph all the stops made are displayed. The other apps obviously use a lower predefined speed value for import and lose the shortest stops)

I guess these apps do something like this:

Movement_time = 0

For each segment in imported_track do

Segment_speed = Segment_length / Segment_time

If Segment_speed < Min_movement_speed
 Then Movement_time = Movement_time + Segment_time

Perhaps it would be good to do something similar for importing into OpenTracks and so it would not be necessary to save additional points without respecting the distance interval.

dennisguse, 2021-10-10:
Besides https://github.com/OpenTracksApp/OpenTracks/issues/968#issuecomment-939537053, OpenTracks does create new segments if:
* pause/resume triggered by the user
* or distance to previous TrackPoint > max distance
If the user remains stationary for some time, nothing happens.
To take care about proper statistics, the additional TrackPoints are stored.

vlmendz, 2021-10-12:
Sounds good to me 👌

Regard to import, I will try to prepare an example. In my opinion, the import feature in OpenTracks is not that important (in fact I would not have used it if there were not the problems in the recording for which opened this issue) but since an error appeared it is worth fixing.

commit 4372f3b1 (2021-10-13): GPS data recording: remove idle/non-idle changes due to reported speed greater than 0.5mph.

This avoids that to storing GPS locations if the user r actually is not moving (likely bad GPS reception).

Fixes #968.
commit 738e64cf (2021-10-16): GPS data recording: remove idle/non-idle changes due to reported speed greater than 0.5mph.

This avoids that to storing GPS locations if the user r actually is not moving (likely bad GPS reception).

Fixes #968.