OpenTracksApp/OpenTracks #969: Sensor data not sometimes ignored (when using GPS in addition)

dennisguse, 2021-10-03:
Yesterday, I participated in a cycling event with known elevation gain and distance.
As I had the impression that the sensor data handling is dropping data (initial assumption distance only), I started OpenTracks on the same smartphone twice (Fdroid version + most recent build from main; and Komoot for verification).

Used sensor: Garmin Bike Sensor 2

Observations:
* (only GPS)  altitude gain/loss only 80% of the one reported by Komoot; distance identical
* (cycling sensor + GPS) distance only 60%, but GPS track looks okay
* (cycling sensor + GPS)  gain only 20% while loss 75%

Best guess: the problem is that we reset the sensor data every time a new GPS location is provided, but we might not store it as the distance to the previous location is not large enough leading to dropped data. Reset should only take  place if we actually stored the data.

Code: 
```
    public void onNewTrackPoint(TrackPoint trackPoint, Distance thresholdHorizontalAccuracy) {
        fillAndReset(trackPoint);

        service.newTrackPoint(trackPoint, thresholdHorizontalAccuracy);
    }
```

I need to create a test case for this.
This issue is presented since we introduced altitude gain/loss and distance sensors.

rgmf, 2021-10-03:
>     * (only GPS)  altitude gain/loss only 80% of the one reported by Komoot; distance identical
> 
>     * (cycling sensor + GPS) distance only 60%, but GPS track looks okay
> 
>     * (cycling sensor + GPS)  gain only 20% while loss 75%

Does cycling sensor have an impact on barometric sensor (gain and loss)?

dennisguse, 2021-10-03:
@rgmf Yes, the data provided by the cycling sensor determines if a TrackPoint is stored or not (min recording distance).
And we always reset the sensor data to 0 ignoring if the data was stored.

commit dc63c753 (2021-10-07): TrackRecordingManager: do not store sensor distance twice (previous TrackPoint's storage was delayed).

Fixes #969.
commit f370c120 (2021-10-08): TrackRecordingManager: do not store altitude changes twice (previous …

…TrackPoint's storage was delayed).

Fixes #969.
commit 10524aa0 (2021-10-13): TrackRecordingManager: do not store altitude changes twice (previous …

…TrackPoint's storage was delayed).

Fixes #969.
commit 1df8e9dc (2021-10-13): Bugfix: reset on SensorDataSet should only take place if the TrackPoint was stored

Fixes #969.